MrChromebox/firmware #176: Acer Spin 13 can't sleep/hibernate on Linux

CiriousJoker, 2019-11-26:
When attempting to do it anyway, the system either shuts down (when on battery) or the fans start spinning at 100% until you resume.

Some folks have said it's a tpm issue, bug apparently, TPM is disabled in the bios already.

Here's a post I wrote about this in the arch forums. There are no replies, but I've uploaded some logs there if you're interested:
https://bbs.archlinux.org/viewtopic.php?id=250514

MrChromebox, 2019-11-27:
what firmware type / version are you running?

CiriousJoker, 2019-11-27:
I'm running on 4.10 remix. The issue happens on UEFI as well as on legacy bios. On legacy bios, the boot flags or smth like that are also reset. This means that after the system is shut down like this (not via normal shut down), the white dev screen comes and when I try to boot from USB, it beeps and that's it. I can only boot Chrome OS at that point. If I recall correctly, before that, the "chrome is is dead" boot screen appears but after a restart it's like I described.

MrChromebox, 2019-11-27:
suspend/resume issues on the legacy BIOS are a product of Google's use of the TPM in their verified boot -- there's nothing I can do about that, it's an issue with the stock firmware.

with the UEFI firmware, I'd need to see the output of the EC serial console to see what's going on. Also highly recommended to update to the 4.11 release to avoid NVRAM issues

CiriousJoker, 2019-11-27:
Is there anything I can do to provide you with that output?

MrChromebox, 2019-11-27:
if you have a USB-C debug cable, log the EC console starting just prior to suspend, let it sit 30s or so, then wake up. That would help me see what's going on with the EC that the fan state isn't getting set properly.

CiriousJoker, 2019-11-27:
Currently, I don't have one, but I found this thing on eBay:
https://www.ebay.it/ulk/itm/122662276987

Would this work?

CiriousJoker, 2019-11-27:
Nevermind, this probably won't work, since the Chromium website says that to access CCD, you need a SuzyQ cable which I obviously don't own and also can't purchase since it's out of stock and relatively expensive for a one time use with uncertain outcome. :(

MrChromebox, 2019-11-27:
Sparkfun shows backordered with 12/9 ETA: https://www.sparkfun.com/products/14746

CiriousJoker, 2019-11-27:
Hmm, maybe I'll do it. If I get one, how would I read the EC console? I can use Windows or Arch Linux. I won't need any extra devices right?

MrChromebox, 2019-11-27:
minicom under Linux is all you'd need. I believe the EC console is readable by default, so you'd just need to connect the cable (correct port, orientation) start minicom on the proper port (-D /dev/ttyUSB1, IIRC), and start logging

empathicqubit, 2020-02-25:
@CiriousJoker Were you able to make any progress on this? I'm running into this issue as well, and I was already considering the SuzyQ in case the flash broke. I ended up just partially unlocking the CCD and crossing my fingers really hard, but I'd still buy it if it meant finding a resolution.

CiriousJoker, 2020-02-25:
No, I was and am super busy with other stuff, sorry :/

empathicqubit, 2020-04-12:
I have the SuzyQable, if someone could walk me through what to do.

empathicqubit, 2020-04-12:
Relevant question regarding the TPM. If I flash my EFI will I lose my keys? The reason I'm asking is because I'm going to make a full image backup, but if I lose the system key then that will make that useless since the cryptohome won't decrypt without it. I created individual backups of Android and Crostini which should cover my bases anyway, but I'd rather not deal with those because I'll still need to reconfigure some stuff again if I have to use them.

MrChromebox, 2020-04-12:
you'll lose the keys if you open the CCD as part of disabling the firmware write protect, since that will transition you out of Developer Mode and will wipe the TPM.

empathicqubit, 2020-04-12:
Thanks, I just discovered that :/

IRT the fan state, I was seeing some messages repeating over and over, "Fan 0 stalled", when suspending with ChromeOS at the first time setup screen. I don't think the I haven't been able to get them to come up again since setting up the system and unlocking the CCD again.

I've attached the suspend logs from ChromeOS with the default firmware, for both CCD and EC ports. I'll do the same for Linux once I go through reflashing.

[sleep_chromeos_0.bin.gz](https://github.com/MrChromebox/firmware/files/4466738/sleep_chromeos_0.bin.gz)

[sleep_chromeos.bin.gz](https://github.com/MrChromebox/firmware/files/4466737/sleep_chromeos.bin.gz)

empathicqubit, 2020-04-12:
Was there a specific kernel I should be using? I'm using Xubuntu 19.10 with 5.3.0-46-generic.

empathicqubit, 2020-04-12:
Linux EC and CCD logs. The suspend was executed with `systemctl suspend` because it wasn't clear to me if the lid switch was actually doing the correct thing.

[sleep_linux_ec.bin.gz](https://github.com/MrChromebox/firmware/files/4467155/sleep_linux_ec.bin.gz)
[sleep_linux_ccd.bin.gz](https://github.com/MrChromebox/firmware/files/4467156/sleep_linux_ccd.bin.gz)

empathicqubit, 2020-04-20:
Would you be able to walk me through how you would use this information to debug the issue? I'm fairly technical and I want to help you. Also I would like to put CrOS back on my machine soon, but I've been putting it off in case you needed me to do something else.

CiriousJoker, 2020-04-20:
@MrChromebox I'm also still interested in this, in case there's a fix, I'll try it out immediately

MrChromebox, 2020-04-21:
having looked at the logs, it's not providing me with any info as to why sleep would be failing, or why the fan wouldn't be resuming coming out of suspend. And I'm not seeing this problem on other (non-ChromeOS) KBL-U boards here

empathicqubit, 2020-04-25:
@MrChromebox Is it possible to do gdb debugging over the Cr50? Would that give you any useful information?

MrChromebox, 2020-04-25:
@empathicqubit I'm not sure, but I don't think so. one can query the fan state from the EC console though, so one could check the fan mode, RPM, duty cycle, etc before and after suspend

empathicqubit, 2020-04-25:
I don't think it's actually suspending so I'm not sure how useful that is, but I'll take a look. I also ran through all the commands in the EC and AP and got an output for that session. I'm not sure if that's useful.

[diag.bin.gz](https://github.com/MrChromebox/firmware/files/4534360/diag.bin.gz)
[diag2.bin.gz](https://github.com/MrChromebox/firmware/files/4534361/diag2.bin.gz)

empathicqubit, 2020-04-26:
Should the UART console be enabled and reporting over ttyUSB1, or am I getting that mixed up? Would that be useful?

MrChromebox, 2020-04-27:
the CPU UART (ttyUSB1) might show the coreboot log on resume, assuming it actually suspends and then tries to resume. Otherwise it's not going to show anything useful

empathicqubit, 2020-04-28:
I built coreboot myself and was able to enable a bunch of messages on the UART console as well as GDB. I was able to load the symbols and step through a bit. Probably useless though as it won't help determine what the correct suspend procedure is. I was thinking about trying to enable DCI in the ChromeOS IFD so that I could trace that instead and get more insight on what it's doing during suspend, but I'm not sure if that's even possible. I'm probably overcomplicating things.

What about the UART console for the ChromeOS kernel? There's a blank console option for it but I'm not sure if setting it will even do anything because documentation seems to imply that support needs to be compiled in and usually isn't.

Sorry for all the nonsense. I have no idea what I'm doing.

MrChromebox, 2020-04-28:
>  Probably useless though as it won't help determine what the correct suspend procedure is

every other SKL/KBL coreboot device I have access to suspends/resumes correctly, I'd find it very surprising if one board is having an issue, vs a kernel or driver issue

empathicqubit, 2020-04-28:
Should I close this bug and go complain to kernel devs?

MrChromebox, 2020-04-28:
I'd say leave it open in case, but definitely pursue the other route as well. I'll see if I can get another NAMI-based board for testing

empathicqubit, 2020-04-29:
Okay. I'll message them soon but I'm going to try experimenting a bit more before bothering them + double-checking to make sure it isn't already fixed in mainline. Does your build contain the blobs for the audio? When I tried building with that option myself it failed because it wasn't in the standard blobs repo. If you're not including that, could it cause it to behave weirdly on suspend?

MrChromebox, 2020-04-29:
> Does your build contain the blobs for the audio?

of course :)

>  When I tried building with that option myself it failed because it wasn't in the standard blobs repo

you extract them from the stock firmware image using cbfstool

empathicqubit, 2020-04-29:
Hmm

```  
TPM Interface Specification 1.3 Interface / TPM 2.0 FIFO Interface - (SPI) (TCG_TIS_SPI) [M/n/y/?] m
    Cr50 SPI Interface (TCG_TIS_SPI_CR50) [N/y/?] (NEW) ?

CONFIG_TCG_TIS_SPI_CR50:

If you have a H1 secure module running Cr50 firmware on SPI bus,
say Yes and it will be accessible from within Linux.

Symbol: TCG_TIS_SPI_CR50 [=n]
Type  : bool
Defined at drivers/char/tpm/Kconfig:70
  Prompt: Cr50 SPI Interface
  Depends on: TCG_TPM [=y] && TCG_TIS_SPI [=m]
  Location:
    -> Device Drivers
      -> Character devices
        -> TPM Hardware Support (TCG_TPM [=y])
          -> TPM Interface Specification 1.3 Interface / TPM 2.0 FIFO Interface - (SPI) (TCG_TIS_SPI [=m])

    Cr50 SPI Interface (TCG_TIS_SPI_CR50) [N/y/?] (NEW) 
```

MrChromebox, 2020-04-30:
that's not affecting suspend on my UEFI firmware, but might help with suspend/resume on the stock firmware

empathicqubit, 2020-04-30:
The good news is that mainline doesn't have the fan issue. The bad news is that suspend freezes the computer. Power button, lid, keyboard won't wake it. It's possible I got something wrong with `make oldconfig` so I'll try working through it again, or maybe trying stable instead.

empathicqubit, 2020-04-30:
Why is it so hard for manufacturers to do this in a standard way? With nearly every Linux laptop I've had this has been a issue at some point.

empathicqubit, 2020-06-08:
Bug filed at https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1882543 . Apologies for the delay. Was trying to figure out the best place to do it. Linux kernel support documentation is not very clear, and Ubuntu already has the data collector so I just decided to use that since it seemed easier. They will probably kick it upstream but I figured it was best for me not to bother them directly.

empathicqubit, 2020-06-08:
Please subscribe to the bug to bump up the score also.

CiriousJoker, 2020-07-24:
@empathicqubit Can you update the launchpad issue? They asked for the `journalctl -b -1 -k` log and since you're the bug creator I doubt they'll accept mine

empathicqubit, 2020-07-24:
I'll try. I saw it but I got back to needing the machine again. Thankfully I didn't have to powerwash the last time I switched the firmware back so I'll try it again soon.

krrrak, 2023-05-11:
same problem in coreboot4.19

MrChromebox, 2023-05-11:
nobody has provided any info which leads me to believe there is a firmware issue

liujustin604, 2025-04-25:
> suspend/resume issues on the legacy BIOS are a product of Google's use of the TPM in their verified boot -- there's nothing I can do about that, it's an issue with the stock firmware.
>
> with the UEFI [...]
<details>
(cr) liujustin604@fedora /mnt/host/source/src/scripts $ sudo usb_console -d 18d1:5014 -i 2
C1 st34
C1 st35
C1 HARD RST TX
[295.486398 mux_state = 0x1, mux_type = 0x1
]
C1 st4
[295.916603 USB MUX 0]
C1 st34
C1 st35
C1 HARD RST TX
[296.172510 mux_state = 0x1, mux_type = 0x1
]
C1 st4
[296.589492 C1: AUX_SW_SEL=0xc]
C1 st5
[298.021038 HC 0x103]
[298.022659 HC 0x113]
[298.023347 HC 0x103]
[298.024198 HC 0x113]
[298.024819 HC 0x103]
[298.025506 HC 0x113]
[575.983389 HC 0xa9]
[576.054087 KB disable]
[576.054373 KB IRQ disable]
[576.055059 KB IRQ enable]
[576.055369 KB enable]
[576.535257 HC 0xa4]
+[576.536296 HC 0x67]
[576.536625 HC 0x67 err 9]
[576.747302 power state 3 = S0, in 0x003d]
[576.747930 power state 8 = S0->S3, in 0x003d]
[576.749932 PD:S0->S3]
[576.750448 power state 2 = S3, in 0x003d]
[576.753606 Lid Accel ODR: 10000 - roundup 1 from config 2 [AP 0]]
[576.758039 Base Accel ODR: 10000 - roundup 1 from config 2 [AP 0]]
[576.765355 Base Gyro ODR: 0 - roundup 0 from config 0 [AP 0]]
Port 80 writes:
  05 25 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 <--new
[714.681701 power state 2 = S3, in 0x003f]
[714.682347 power state 7 = S3->S0, in 0x003f]
[714.684591 PD:S3->S0]
[714.686800 power state 3 = S0, in 0x003f]
[714.696533 Lid Accel ODR: 10000 - roundup 1 from config 1 [AP 0]]
[714.709465 Base Accel ODR: 10000 - roundup 1 from config 1 [AP 0]]
[714.717357 Base Gyro ODR: 0 - roundup 0 from config 0 [AP 0]]
[714.953682 HC 0x11f]
+[714.979465 HC 0x0d]
[714.980012 HC 0xa4]
++[714.983548 HC 0x121]
[714.983925 HC 0x121 err 1]
[714.984414 HC 0x02]
+[715.116420 KB disable]
[715.116744 KB IRQ disable]
[715.117180 KB IRQ enable]
[715.117490 KB enable]
[715.117823 KS disable]
[715.119479 KS enable]
[715.213433 HC 0xa9]
[715.214040 HC 0x67]
[715.214370 HC 0x67 err 9]
[715.215056 HC 0x103]
[715.216685 HC 0x113]
[715.221481 HC 0x103]
[715.222168 HC 0x113]
Port 80 writes:
  05 25 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
(S3->S0)
  00 02 7f 00 00 0f 03 23 48 49 4a 4d 50 15 52 55 20 2f 4f 5f
  0a 6f 26 7f 00 06 08 0a 1f 20 3f 40 50 52 7f 00 1c 1e 21 22
  23 24 43 70 71 61 5a 5d 1c 1e 21 22 23 24 43 70 71 61 5a 5d
  01 21 55 7f 00 00 09 00 02 03 04 05 06 08 0a 0b 0c 0d 07 0e
  13 14 15 16 19 7f 00 01 02 03 14 16 18 20 30 32 4f 50 5f 15
  20 21 22 23 24 28 3f 4f 61 63 64 6f 40 41 42 4d 47 70 7f 00
  00 00 00 <--new
</details>

liujustin604, 2025-04-25:
(cr) liujustin604@fedora /mnt/host/source/src/scripts $ sudo usb_console -d 18d1:5014 -i 2
C1 st34
C1 st35
C1 HARD RST TX
[295.486398 mux_state = 0x1, mux_type = 0x1
]
C1 st4
[295.916603 USB MUX 0]
C1 st34
C1 st35
C1 HARD RST TX
[296.172510 mux_state = 0x1, mux_type = 0x1
]
C1 st4
[296.589492 C1: AUX_SW_SEL=0xc]
C1 st5
[298.021038 HC 0x103]
[298.022659 HC 0x113]
[298.023347 HC 0x103]
[298.024198 HC 0x113]
[298.024819 HC 0x103]
[298.025506 HC 0x113]
[575.983389 HC 0xa9]
[576.054087 KB disable]
[576.054373 KB IRQ disable]
[576.055059 KB IRQ enable]
[576.055369 KB enable]
[576.535257 HC 0xa4]
+[576.536296 HC 0x67]
[576.536625 HC 0x67 err 9]
[576.747302 power state 3 = S0, in 0x003d]
[576.747930 power state 8 = S0->S3, in 0x003d]
[576.749932 PD:S0->S3]
[576.750448 power state 2 = S3, in 0x003d]
[576.753606 Lid Accel ODR: 10000 - roundup 1 from config 2 [AP 0]]
[576.758039 Base Accel ODR: 10000 - roundup 1 from config 2 [AP 0]]
[576.765355 Base Gyro ODR: 0 - roundup 0 from config 0 [AP 0]]
Port 80 writes:
  05 25 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 <--new
[714.681701 power state 2 = S3, in 0x003f]
[714.682347 power state 7 = S3->S0, in 0x003f]
[714.684591 PD:S3->S0]
[714.686800 power state 3 = S0, in 0x003f]
[714.696533 Lid Accel ODR: 10000 - roundup 1 from config 1 [AP 0]]
[714.709465 Base Accel ODR: 10000 - roundup 1 from config 1 [AP 0]]
[714.717357 Base Gyro ODR: 0 - roundup 0 from config 0 [AP 0]]
[714.953682 HC 0x11f]
+[714.979465 HC 0x0d]
[714.980012 HC 0xa4]
++[714.983548 HC 0x121]
[714.983925 HC 0x121 err 1]
[714.984414 HC 0x02]
+[715.116420 KB disable]
[715.116744 KB IRQ disable]
[715.117180 KB IRQ enable]
[715.117490 KB enable]
[715.117823 KS disable]
[715.119479 KS enable]
[715.213433 HC 0xa9]
[715.214040 HC 0x67]
[715.214370 HC 0x67 err 9]
[715.215056 HC 0x103]
[715.216685 HC 0x113]
[715.221481 HC 0x103]
[715.222168 HC 0x113]
Port 80 writes:
  05 25 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
(S3->S0)
  00 02 7f 00 00 0f 03 23 48 49 4a 4d 50 15 52 55 20 2f 4f 5f
  0a 6f 26 7f 00 06 08 0a 1f 20 3f 40 50 52 7f 00 1c 1e 21 22
  23 24 43 70 71 61 5a 5d 1c 1e 21 22 23 24 43 70 71 61 5a 5d
  01 21 55 7f 00 00 09 00 02 03 04 05 06 08 0a 0b 0c 0d 07 0e
  13 14 15 16 19 7f 00 01 02 03 14 16 18 20 30 32 4f 50 5f 15
  20 21 22 23 24 28 3f 4f 61 63 64 6f 40 41 42 4d 47 70 7f 00
  00 00 00 <--new

WirelessFidelity1, 2025-04-26:
@liujustin604 , thanks for sharing. Would you be able to provide more information on the device's behavior when attempting to suspend? Was a previous version of the UEFI firmware working properly with suspend? I have tested with Fedora 42 as well, and in my case the system will suspend, but upon opening the lid fans go to max and hard restart. 

Also curious if you have tried switching between `s2idle` and `deep` suspend modes on your system. I wasn't able to get either to work entirely on my CP713-1WN with the i3-8130u.

ronan615, 2025-05-04:
> [@liujustin604](https://github.com/liujustin604) , thanks for sharing. Would you be able to provide more information on the device's behavior when attempting to suspend? Was a previous version of [...]

I have the same cpu. I was unable to get s2idle and deep suspend modes to work. The fan just started to spin really fast or it just shut down. I was using the latest firmware but it did not appear to work on any versions.

WirelessFidelity1, 2025-06-10:
I've just ordered a suzyq cable to do some more debugging on this bug. Once I get it in I'll report back with my findings. I'd really like to use this as a fedora device but the sleep bug makes it really inconvenient for daily usage.

WirelessFidelity1, 2025-08-20:
I never did figure out what the issue was. I used a suzyq but just got the exact messages posted above, which isn't too telling toward the sleep bug. Closest indication I got was 0x67 err 9, which appears to just be a periodic normal message. I got another laptop that works properly so gave up on this one. Sad update but laptop isn't too useful for me without a proper sleep mode. I wouldn't get this laptop if you want to run anything besides chrome os and still use it portably.

sh0dZz, 2025-09-02:
Hey. Hibernate works properly for me on Manjaro with Gnome. I never managed to get suspend to work but hibernate works properly. After powering the system back on its still in the state it was in. The apps remain in the same state. It does require making a swap partition but for me its a great way to replace suspend anyway as it uses less energy. My Spin 13 model is CP713-1WN.