MrChromebox/firmware #177: Shutdown problem

WeirdTreeThing, 2019-11-27:
I installed the RW_Legacy firmwae on my Careena chomebook and when I held the power button to reboot the power menu didn't appear so i pressed the power button in the status area and it went to a white screen and didn't do anything so i forced a shutdown and it booted normally. Then I tried to shutdown again and it worked properly. I thought the firmware didn't install properly so I installed it again and the same thing happened but this time I typed sudo poweroff in the terminal and it worked the right way and I rebooted again and it worked. I don't think this is a big problem but it might break stuff in the future but for me it is a little annoying.

MrChromebox, 2019-11-27:
the RW_LEGACY firmware has no ability to affect anything when you've booted ChromeOS. Any issues you're having in ChromeOS are unrelated to the RW_LEGACY update

WeirdTreeThing, 2019-11-27:
OK i just thought it was weird that it did that

WeirdTreeThing, 2019-11-28:
Maybe the script does something to chrome os causing the temporary issue

MrChromebox, 2019-11-28:
it's really not possible. The script just sets a flag and writes the RW_LEGACY part of the firmware. That part is only executed when you hit CTRL+L on the dev mode boot screen. If you boot ChromeOS, it's like that part doesn't even exist

WeirdTreeThing, 2019-11-28:
Ok it might just be a bug on my device only

closed 2019-11-28 by WeirdTreeThing