MrChromebox/firmware #178: Firmware Request AMD APUs

platinumdubs, 2019-11-29:
I know AMD isn't the biggest market when it comes to Chromebooks there's only a few. messing around with chromefy and grunt recovery i can get my HP 2000 notebook has a crappy e-300 apu but i can boot to the Chrome OS splash screen but then hangs there and i have to hard reboot. no worries if you don't have time i'm in the process of downloading these source codes gonna try and make a build that'll work with this older AMD apu however this'll be my first time attempting to build anything from source so i don't expect to be done tomorrow lol. if you got any ideas let me know thanks in advance.

MrChromebox, 2019-11-29:
I have some AMD Chromebooks here for testing, but currently they don't boot when built from upstream coreboot so need to sort that out.  Not sure how that relates to your trying to hack ChromeOS onto an older AMD laptop though.

platinumdubs, 2019-11-29:
i mean id prefer to have a build that works rather than one thats frankensteined together lol. i just wasnt sure if you had one already in the works and if it would be compatible with the apu of this laptop or not. im sure you can prolly get one together faster than im going to be able to lol

MrChromebox, 2019-11-29:
you seem to be misunderstanding what I provide -- I build device firmware to allow Chromebooks to run Linux/Windows, not ChromeOS images for non-Chromebooks.

platinumdubs, 2019-11-29:
no i get that but figured there wasnt any harm in askin..

closed 2019-11-29 by MrChromebox