MrChromebox/firmware #181: Elan touchpad and touchscreen not working - Dell 3189 /(KEFKA) Braswell + MrChromebox-4.11.1 + Gallium 3.0

twisteros, 2019-12-05:
I had Gallium 2.x and then 3.0 running on built-in firmware. After some time accidentally 
reset the bootloader flag and had to re-install Chrome OS. 

Instead of setting the cross system flag I installed the latest MrChromebox firmware. 
When I launched Gallium off the memory stick it turned out that the touchpad and touchscreen
don't work.

pcs3rd, 2019-12-19:
@twisteros I have a similar issue with just the touchscreen, may help if you provide same things in issue #172 with everything in separate pastebins for him to compare.

MrChromebox, 2019-12-19:
found a few things that might be related, shoot me an email so I can have you try a test build - MrChromebox at gmail

twisteros, 2019-12-21:
@pcs3rd : I've just reverted to legacy boot and Gallium 3.0 started working with no issues. The only pain is to remember about CTRL-L every time I boot the laptop. As a workaround every time when I finish working I put the machine to sleep instead of shutting down.

MrChromebox, 2019-12-21:
@twisteros I have a test build I'd like you to try, shoot me an email

papppmac, 2020-02-12:
I also have a Dell 3189/KEFKA that is flashed with Mr.Chromebox-4.11.1 and running ArchLinux, I am having similar pointing device problems.  I wanted to add some details to the thread while poking at it.  

Both devices seemed to work correctly under factory firmware + ChromeOS before I started tampering, so I believe the hardware is fine. 
I've also booted it with a Galliumos-3.1-braswell image for contrast, all of these behaviors seem to be consistent across distributions and kernel versions. 

The touchscreen is using the elants_i2c module, the clickpad is using elan_i2c.

With either enabled, I get high-speed spray of kernel log messages and partial functionality; 
with only the touchscreen (elants_i2c) module loaded, the log message is 
`elants_i2c i2c-ELAN0001:00: unknown packet ff ff ff ff`
with only the touchpad (elan_i2c) module loaded, the log message is
`elan_i2c i2c-ELAN0000:00: invalid report id data (ff)`
 touching the device whose module isn't loaded seems to set off the spray of log messages.

If both modules load, one device sort-of works (inconsistent movement, log filled with above errors) and the other is inert, which one works depends on some factor I haven't figured out, it seemed to randomize on reboot so I'm guessing a timing thing.

The only place I see the strings for those errors in the kernel modules is in the interrupt handler functions, so I'd strongly guess an IRQ problem. 

I'm having sound card problems as well and see a 
```
rt5640 i2c-10EC5650:00: Device with ID reigster 0x6419 is not rt5640/39
rt5645 i2c-10EC5650:00: i2c-10EC5650:00 supply avdd not found, using dummy regulator
rt5645 i2c-10EC5650:00: i2c-10EC5650:00 supply cpvdd not found, using dummy regulator
```
in the kernel log, which also looks like (potentially related) something not being initialized correctly around the i2c bus and associated interrupts. 

I picked up this machine just to hack on for fun, so if you want a guinea pig for anything, I'm game.

papppmac, 2020-02-13:
I got a minute to poke at it this evening, flashed the machine back to stock+legacyrw and booted from an Arch USB.  
Neither touch device causes errors on stock firmware.  

The IRQ map looks ...significantly... different on the two firmwares.  Most relevant for this situation, there are two elan devices with interrupts mapped at 183 and 184 on the legacyrw boot, and only a single elan device at 183 on the uefi boot. 

Here's the contents of /proc/interrupts to compare the map from boots on the different firmwares. 
[interrupts.legacyrw.txt](https://github.com/MrChromebox/firmware/files/4196424/interrupts.legacyrw.txt)
[interrupts.mrchromebox.txt](https://github.com/MrChromebox/firmware/files/4196427/interrupts.mrchromebox.txt)

And here are the two boots' kernel logs for comparison:  [dmesg.legacyrw.txt](https://github.com/MrChromebox/firmware/files/4196418/dmesg.legacyrw.txt) and [dmesg.mrchromebox.txt](https://github.com/MrChromebox/firmware/files/4196420/dmesg.mrchromebox.txt).  
I'm not entirely sure anymore if the elan_i2c and elants_i2c modules were blacklisted during the uefi boot. 

This isn't a part of the boot process I'm knowledgeable about so I'm not quite sure what to look for. 
All attachments appended with .txt so I can lazily make github host them. 
I can run whatever other tests might be helpful (...modulo whatever it is I broke when I installed a bios bootloader over an EFI bootloader on an existing install), the machine is a toy to hack on.

MrChromebox, 2020-02-13:
the `dmesg.mrchromebox.txt` file appears to have the elan drivers backlisted as I don't see anything there. Interesting that an interupt isn't getting assigned to the touchscreen. All the rest seem sane

closed 2022-08-24 by MrChromebox