MrChromebox/firmware #182: System suspend issue under Ubuntu 19.10 Google Pixel 2013

kukusyaoran, 2019-12-16:
With Ubuntu 19.10 when the lib off the system doesn't suspend, the light bar keep still on.
System report show something about cros-ec-lightbar:
[image]
This issue only happens on Ubuntu 19.10, across coreboot 4.10-4.11.1, with Ubuntu 19.04 it performs normal to suspend.

MrChromebox, 2019-12-16:
is there any reason to suspect a firmware issue and not a kernel regression?

kukusyaoran, 2019-12-16:
> is there any reason to suspect a firmware issue and not a kernel regression?

Yes, you'r right. I also suspect the kernel but didn't test it. It's normal on kernel version 5.0.0 but 5.3.0 which not. I'll try figure it out in more detail.

whmzsu, 2020-03-09:
Same problem here. So I  switch back to the ppa mainline  kernele 5.0   and the suspend problem gone

williamyoon, 2020-04-24:
Hi, I am completely new to linux and pixel2013+chromebox.  Can you point me to a guide on how to "switch back to the ppa mainline kernele 5.0" ?

whmzsu, 2020-04-24:
see the wiki   https://wiki.ubuntu.com/Kernel/MainlineBuilds

MrChromebox, 2020-04-24:
this really isn't the right venue for OS/kernel issues/support. There doesn't appear to be any firmware-level issue causing issues with suspend on newer kernels, so closing this

closed 2020-04-24 by MrChromebox