MrChromebox/firmware #183: Linux kernel module chromeos_tbmc broken by firmware?

delipunch, 2019-12-19:
Linux kernel module `chromeos_tbmc` doesn't put out acpi events when lid is opened beyond 180 degrees, possibly because of the firmware changing the `goog0006` device to appear as its Intel counterpart.

MrChromebox, 2019-12-19:
on what device(s)?

delipunch, 2019-12-19:
So far, asus c302c cave and 'Dell Chromebook 11 3189 (2 in 1) braswell' as per this survey I conducted https://www.reddit.com/r/GalliumOS/comments/e67bec/calling_all_2_in_1_owners_please_perform_this_test/

MrChromebox, 2019-12-19:
well, the whole point of switching to the Intel ACPI handler was that the ChromeOS driver wouldn't need to use used (and wouldn't need to be ported to other OSes which already support the Intel one)

delipunch, 2019-12-19:
Is the appropriate module `intel-vbtn`, then? If so, that doesn't do anything that I can see when I check with `acpi_listen`, `xev`, or `dmesg`. I see in the [source for intel-vbtn](https://github.com/torvalds/linux/blob/master/drivers/platform/x86/intel-vbtn.c) that it looks for the `INT33D6` device that `GOOG0006` is mapped to. (Side question; would have compiling `chromeos_tbmc.c` with `INT33D6 instead of `GOOG0006` changed anything?)

MrChromebox, 2019-12-20:
I honestly haven't tested much at all under Linux. The switch to VBTN from TBMC was part of getting the Pixelbook fully functional under Windows. If it's problematic then I'll look into how I can handle both use cases -- possible the two can coexist.

delipunch, 2020-06-19:
Looking through the github links for firmware release 4.12, I saw this commit (https://github.com/MrChromebox/coreboot/commit/48ab1e4a2c9395e59d91ac02ae1e6c23c46c20a5) that seemed to indicate that this issue might've been fixed.  I gave the new firmware a quick test, and it didn't change from last time - no events whatsoever when laptop lid was beyond 180 degrees.  I'll do more extensive tests with different kernel versions soon.

MrChromebox, 2020-06-19:
@delipunch you linked to the fix but didn't read it. You need to add the kernel param as listed in order for the TBMC part to be used, otherwise it uses VBTN

delipunch, 2020-06-19:
I did add the kernel parameter, I just thought reading the commit and implementing the kernel parameters was assumed to of happened. `uname -r` returns `5.3.0-050300-generic`, still doesn't work with `chromeos_tbmc` or `intel-vbtn`

benklop, 2022-01-04:
this still seems to be a problem on skylake/caroline with 5.13.19, firmware 4.12-MrChromebox, both with and without acpi_osi=Linux as a kernel parameter in grub.

benklop, 2022-01-04:
udevadm info --export-db|grep GOOG0006 shows nothing, so I'm pretty sure that GOOG0006 is not exposed.

MrChromebox, 2022-01-04:
well 4.12 is almost 2 years old, so I'd try again with 4.14 (my 4.15-based release will hopefully be out end of this month) as that will expose both the google and Intel tablet mode devices

benklop, 2022-01-04:
I'm sorry, i made a typo. It is in fact running MrChromebox-4.14 07/25/2021

benklop, 2022-01-04:
So if i understand correctly, the current firmware I am using, 4.14, should expose the GOOG0006 device. Since I am not seeing it (nor am I seeing the intel flavored INT33D6 device), does that mean caroline does not expose this tablet mode ACPI from the EC and uses some other method under ChromeOS to detect tablet mode?

benklop, 2022-01-04:
https://github.com/MrChromebox/coreboot/blob/2021.07.25/src/mainboard/google/glados/variants/caroline/include/variant/ec.h definitely does not enable this feature, so that's why it's not included. I'm going to try to enable it and build the firmware.

MrChromebox, 2022-01-04:
that's not going to do anything without the accompanying change to the EC firmware, and likely some code to actually make use of the driver

benklop, 2022-01-04:
https://github.com/MrChromebox/chrome-ec/blob/master/board/glados/board.h appears to include CONFIG_LID_ANGLE, so it seems to me that the ec firmware should already support it. I'm not sure how to check what version the ec firmware in my chromebook is at, however.

benklop, 2022-01-04:
as far as code to make use of the driver, I will worry about that when the device is presented to Linux. I believe, however, that if it is presented as an acpi tablet switch then libinput should be able to use it automatically.

benklop, 2022-01-06:
Success! After the challenge of building the firmware, adding the following patch, ACPI events for entering and exiting tablet mode are reported by the OS. The ec firmware for glados must support the feature then :)

```
diff --git a/src/mainboard/google/glados/variants/caroline/include/variant/ec.h b/src/mainboard/google/glados/variants/caroline/include/variant/ec.h
index d0001b4b12..610507ff86 100644
--- a/src/mainboard/google/glados/variants/caroline/include/variant/ec.h
+++ b/src/mainboard/google/glados/variants/caroline/include/variant/ec.h
@@ -1,4 +1,7 @@
 /* SPDX-License-Identifier: GPL-2.0-only */
 
+/* Enable Tablet switch */
+#define EC_ENABLE_TBMC_DEVICE
+
 /* EC ENABLE MULTIPLE DPTF PROFILES */
 #define EC_ENABLE_MULTIPLE_DPTF_PROFILES
```

benklop, 2022-01-06:
This should be possible to enable for any glados based chromebook with a tablet mode. i have no idea which ones that would include, but certainly caroline.

benklop, 2022-01-06:
As far as software to use the feature, I can now confirm that KDE Wayland uses libinput, and detects tablet mode correctly, disabling and enabling screen rotation. libinput itself disables and enables the keyboard and trackpad automatically as well.

MrChromebox, 2022-01-06:
I can add this for caroline and cave; the others are not convertible form factor, just regular clamshell

mperreir, 2022-03-26:
Hello,
Could this EC_ENABLE_TBMC_DEVICE fix be also added to cyan board ? The current firmware for cyan already disables the touchpad and keyboard when the device is in tablet mode, so it seems EC_ENABLE_TBMC_DEVICE just has to be added to ec.h (like for caroline and cave). This would allow linux distros to corrreclty enable onscreen keyboard when in tablet mode.

mperreir, 2023-04-26:
Hello @MrChromebox ,
If you have to time to have a look at this (old) request :arrow_up: . I'm still using the device and never took the time to patch and compile myself the firmware for cyan. It's certainly much easier for you to patch and compile (as was done for caroline and cave).

ShapeShifter499, 2023-08-26:
@MrChromebox I have a Samsung Chromebook Plus v2 code name nautilus. I'm noticing that there is a "GOOG0006" device exposed with the name "Tablet mode switch" and libinput is correctly tying the keyboard and trackpad to that device.  But it doesn't send any actual switch events. Instead the "INT33D6" device is sending all of the tablet switch events and that's labeled as "Intel virtual switches".

I tried adding `acpi_osi=Linux` to my kernel command line, systemd-boot  and it didn't seem to change anything.

Do you have any idea what might be happening here?

ShapeShifter499, 2023-09-02:
@MrChromebox should I open a new issue for this?

MrChromebox, 2023-09-02:
@ShapeShifter499 no - in the upcoming 4.21 release, you'll just want to blacklist the Intel tablet button module under Linux

ShapeShifter499, 2023-09-03:
Gotcha, thank you for all of the help!

ShapeShifter499, 2023-09-23:
@MrChromebox any updates on this? Also, do you happen to know the name of the module I would blacklist?

MrChromebox, 2023-09-23:
ran into some issues with getting 4.21 out the door. If you want to try a test build then email or DM me on Discord

MrChromebox, 2023-10-12:
closing this as it should be fixed in 4.21

closed 2023-10-12 by MrChromebox