MrChromebox/firmware #184: Broken sleep with coreboot 4.11 Respin (4.11.1 point release) on Windows

Ke1i, 2019-12-26:
**Hardware details**
-Chromebook model: HP Chromebook (FALCO)
-Operating System: Windows 10 Enterprise LTSC
-Operating System architecture: x64
-Operating System version: 1809

**Problem details**
-Action: Put computer to sleep
-Expected behavior: Computer to sleep/suspend and resume open lid open/mouse click/keyboard
-Actual behavior: Computer boots up like it has been restarted. Meaning it lost power (as if I pulled out the battery, which isn't even removable)

Additional details
-I booted LMDE on USB and I couldn't get the computer to sleep in Linux
-The computer would sleep/suspend normally before the above firmware update

[image]

MrChromebox, 2019-12-26:
how are you performing the suspend/resume? the only way I'm able to generate a failure is to suspend the device manually (ie, via shutdown menu), then close and open the lid. If I suspend via lid close, then it resumes properly on lid open.

Ke1i, 2019-12-26:
Thanks for the fast response.
I perform sleep via the power menu, so manually.
Start>Power Button>Sleep
The below screengrab is from a few minutes ago. Battery charge is at 88%.

[image]

Ke1i, 2019-12-26:
I used DISM to install Windows and that's why it looks like I'm running it from USB. But Windows is running from the SSD drive.

MrChromebox, 2019-12-26:
can you try use closing/opening? just to confirm same behavior I'm seeing here?

I did test the old 4.10 release and not present there, so issue seems to have be introduced in upstream coreboot in between the 4.10 and 4.11 releases

Ke1i, 2019-12-26:
I had disabled the option (sleep on closing the lid) but I've just enabled it and tried a couple of times and it seems to resume just fine upon opening the lid.

MrChromebox, 2019-12-26:
ok, sounds like we're seeing the same behavior then. Only seems to affect older platforms.  I'll chase it down and try to have a fix for the next release

MrChromebox, 2019-12-27:
I've found the issue and have a temp fix in place while I investigate the root cause and proper resolution. Drop me an email MrChromebox at gmail and I'll send you a link to flash a test build.

Ke1i, 2019-12-27:
That was fast!

Ke1i, 2019-12-27:
Thanks for the temporary fix, it works. I've tried a few times and the computer resumes normally from sleep after manual suspend.
Let me continue testing, I'll report back should anything unexpected crop up.

MrChromebox, 2020-03-18:
should be fixed in 4.11.2 release!

closed 2020-03-18 by MrChromebox