MrChromebox/firmware #187: Is there a way to boot from a server hosting the OS image?

wicadmin, 2020-01-02:
I currently do this with Libreelec in a similar fashion (i.e. using iPXE/iscsi from Seabios to retrieve and boot the OS image). These I have within walking distance so in case there is a problem, no a big deal. However, I have some remote servers that I would like to retrieve the OS when it is powered up and boot into that OS. That way, I can update the OS image and remotely power cycle the device. If an issue, I can put back the last working OS image and power cycle again. Is there a way to do something similar to cloud-config of is the iPXE/iscsi way the only way to do something remotely without IPMI or an IPMI dongle (most I see requires a separate monitor for the HDMI on my end, others outdate Java browser plugin etc.)

Thank you.

MrChromebox, 2020-01-02:
I believe there should be an iPXE equivalent for the UEFI firmware, though not currently enabled. I'm not sure how you'd be able to do that automatically/remotely though

Hurricos, 2020-04-11:
Hey wicadmin,

Your best bet, if running SeaBios, is to have it network boot by default, and to use dnsmasq as a TFTP server to serve up an ipxe image embedded with a script.

like so:

```
# /etc/dnsmasq.conf
# ... you also need config to let dnsmasq know which interface to use ...
enable-tftp
tftp-root=/netboot/tftp
pxe-service=x86PC, "Use iPXE",file.unidonly.kpxe
````

... with file.undionly.kpxe being a file under /netboot/tftp containing a build of iPXE [produced like this](https://ipxe.org/download) containing an embedded script which then does what you need.

If you need to fix things at the root, you can rebuild the file.undionly.kpxe on your networking device and change the script without changing the server.

For UEFI: No clue. Due to the Tianocore CorebootPayloadPkg not even supporting the UEFI shell `tftp` command (??!!?), I have not been able to do any kind of netbooting from Tianocore.

MrChromebox, 2025-12-03:
ChromeOS devices with onboard LAN now support iPXE