MrChromebox/firmware #188: Stock chromeos firmware download potentially corrupted.

pcs3rd, 2020-01-19:
I did already send an email to @MrChromebox related to this (That issue was due to a bug in flashrom with the raspi)  
But it seems that whenever I try to restore the download for the chromeos firmware, my chromebook refuses to boot.  
Right now, I have the coreboot uefi firmware installed, but can do reads and whatever, but it may take time because it is a pain in the butt.
[I have included a link to the .rom file](https://drive.google.com/open?id=1hiTg5hueBldtvIyetlkivXExOYUld1bs)

MrChromebox, 2020-01-19:
the shellball downloaded from my site was likely an older version which wasn't new enough for the setzer board you have. I've extracted an updated image and uploaded to my site, so give that a try and see if any better result

pcs3rd, 2020-01-19:
Will do! is there any way to confir it may boot before rebooting

MrChromebox, 2020-01-19:
this shellball is pulled from current Google recovery image for Setzer. If it doesn't boot when flashed externally, there's a serious problem (internal flashing should work too, just saying external better be 100%)

pcs3rd, 2020-01-19:
ok

pcs3rd, 2020-01-19:
@MrChromebox It worked!

pcs3rd, 2020-01-19:
Also @MrChromebox would I be able to flash the eufi firmware into the place where RW+Legacy would go or isthat a no-go?

closed 2020-01-19 by pcs3rd

MrChromebox, 2020-01-19:
newer platforms support running Tianocore as a RW_LEGACY payload, but Braswell isn't one of those last time I tested. Right now, only working on AMD Chromebooks. Should work on Kabylake but doesn't currently