MrChromebox/firmware #191: Auron_Yuna - laptop display broken under windows

igel-itzo, 2020-02-19:
Auron_yuna With windows 8.1 pro Firmware 20191201 (4.11.1) 

As soon as Windows loads the display driver, the integrated laptop display becomes garbled (like a CRT monitor that is out of sync with its scanning and flyback signals). HDMI connected display is not affected. Dowgrading intel drivers to 4322 (very old from 2015) or older was the only way to stabilize the internal display. Current drivers are unusable. Ubuntu, Clearlinux and Fedora do not seem affected.
 
Was running fine with the 20190104 re-release of the 20181230 firmware with the latest intel driver (same that produce display instability with new firmware). 

Tried to dowload the 20190104 firmware but it looks like the is no longer available.

igel-itzo, 2020-02-20:
I tried to clone the 2019.01.04 branch and compile it myself, though it wasn't very clean. The submodules didn't download on their own, as well as I had to make some changes to the extract scripts to get blobs (me, pci video oprom, mrc, refcode, ec)  from the BOOT_STUB potion and fish the GOP driver from a lenovo bios for the same cpu (200-01ibw), though that's a box and not a laptop. I ended up with a rom with the following structure 

```
FMAP REGION: COREBOOT
Name                           Offset     Type           Size   Comp
cbfs master header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           38604 none
cpu_microcode_blob.bin         0x97c0     microcode       70720 none
fallback/ramstage              0x1ac80    stage           93128 none
fallback/refcode               0x31880    stage           75595 none
config                         0x44040    raw              1106 none
revision                       0x44500    raw               590 none
ecrw                           0x447c0    raw            262144 none
ecrw.hash                      0x84800    raw                32 none
vbt.bin                        0x84880    raw             36919 LZMA (65536 decompressed)
cmos_layout.bin                0x8d900    cmos_layout       904 none
fallback/postcar               0x8dcc0    stage           15328 none
fallback/dsdt.aml              0x91900    raw             17690 none
fallback/payload               0x95e80    simple elf     804881 none
spd.bin                        0x15a700   spd              4096 none
(empty)                        0x15b740   null           280152 none
mrc.bin                        0x19fdc0   mrc            223640 none
(empty)                        0x1d67c0   null           167448 none
bootblock                      0x1ff600   bootblock        1968 none
```

it probably passes a sniff test but is it safe to flash?

MrChromebox, 2020-02-20:
flash away as long as you have the ability to unbrick. Or I can push a new build to fix the issue

igel-itzo, 2020-02-20:
My spi clip is nowhere to be found, so need to get a new one before having unbrick capabilities.

igel-itzo, 2020-02-24:
What I compiled was a no-go. Got two blue lights - battery charged and machine on, but no display or backlit, didn't check if HDMI was on. Unbricked, but still having the original problem. 
What would be the most likely issue - blob related or just bad compile?

MrChromebox, 2020-02-24:
I'm afraid I don't have time to try and debug your build, but lack of backlight suggests that one or more of the following are at fault: VBT, GOP driver, or Tiano branch (must use master).

I'll push a new 4.11-1 build later tonight that fixes this for all Broadwell Chromebooks

MrChromebox, 2020-03-18:
fixed in 2020.03.17 release!

closed 2020-03-18 by MrChromebox