MrChromebox/firmware #192: RW_LEGACY on LIARA (Lenovo S345 AMD "Stoneyridge")

ryanwoodsmall, 2020-03-02:
I have a Lenovo S345-14AST Chromebook that I've updated with your RW_LEGACY firmware, but attempting to legacy boot via CTRL-L always shows the black-on-white "Alternative firmware" screen and seems to stop there. I'm wondering if this is the same issue you've seen on BARLA/CAREENA models? Is there anything else I can try on this model that would help?

I've also unlocked the firmware write protect with a SuzyQable, and would be happy to be a guinea pig for full UEFI when/if/before it's ready if you need a tester. I have a cheapy CAREENA model on the way from Woot as well.

Thank you for all your work on Chromebook firmwares!

MrChromebox, 2020-03-02:
looks like they screwed up the text color. Just press 2 for Tianocore and it should load immediately

I have BARLA, KASUMI, and LIARA here so have testing covered, but thanks :)

ryanwoodsmall, 2020-03-02:
Pressing 2 just beeps (with either a USB or MicroSD Ubuntu installer inserted), 1 drops to u-boot. At least a little closer, this gives me something to tinker with.

MrChromebox, 2020-03-02:
2 beeps means that payload slot is invalid / doesn't contain a bootable payload. You updated via my script recently?

MrChromebox, 2020-03-02:
I'll set my LIARA back to stock to re-test if needed, but it worked a few weeks ago

ryanwoodsmall, 2020-03-02:
Yep, updated tonight.

ryanwoodsmall, 2020-03-02:
Ugh, could be a hybrid ISO+UEFI image from Ubuntu. Messing with that now.

MrChromebox, 2020-03-02:
no, you would see the Tianocore boot splash and be able to select the usb from the boot menu regardless of the validity of its contents

ryanwoodsmall, 2020-03-02:
I can manually `load usb 0:1 0x01000000 bios.bin.elf-1.13.0` and `bootelf` from USB storage in u-boot and at least start a stock SeaBIOS payload so I know everything is there. Doesn't seem to react to ESC though, and boots from the eMMC, which of course doesn't work. This is the first Chromebook that's given me any trouble, but I've really only used your UEFI firmware.

MrChromebox, 2020-03-02:
I have no intention of supporting u-boot, no idea why Google added it in.

I just reset my LIARA to stock, flashed the RW_LEGACY update, rebooted, CTRL+L, then 2. Booted right into Tianocore.  ChromeOS stable channel, 79.0.3945.123.  Active firmware Google_Grunt_11031.86.0

ryanwoodsmall, 2020-03-02:
Thanks for verifying. Will give it another shot when I get home from work and double check versions. For grins last evening, I wiped the eMMC, restored Chrome OS from a USB key, reflashed your RW_LEGACY, reset GBB from your script - still only getting beeps upon pressing 2 with a number of different install media. Debian, Ubuntu and Tiny Core all exhibited the same symptoms with `dev_boot_legacy` and `dev_boot_usb` set to **1** and `dev_boot_signed_only` at **0**.

Only other thing I can think of: I have the Best Buy exclusive LIARA w/touchscreen and lighted keyboard. Don't think the hardware is different enough wrt to RW_LEGACY not working here but frankly I'm a bit stumped.

MrChromebox, 2020-03-02:
you don't even need any media to test this, Tianocore will run regardless. there is no need to mess with any crossystem flags - the only one that matters is dev_boot_legacy, and the script sets that for you when flashing the RW_LEGACY firmware

ryanwoodsmall, 2020-03-02:
I dropped out of developer mode, re-enabled dev mode, updated to Chrome OS 80.0.3987.128, reran the latest `firmware-util.sh` to make sure RW_LEGACY was up-to-date, and... it worked. Booted Ubuntu successfully from a microSD. My Chrome OS and firmware versions were the same as yours before the OS update. Very odd. Thanks again for bearing with me, sorry for the wild goose chase.

closed 2020-03-02 by ryanwoodsmall