MrChromebox/firmware #193: Lumpy XE550C22 chromebook touchpad no longer working under linux after update

hkzlab, 2020-03-03:
Hello,
Not really sure if this is the best place to notify this, but the issue appeared right after I updated my firmware image...

Yesterday I took my old 550 out of storage so I could give it to a friend.
It was flashed with an old (January 2019 I think) Full ROM by MrChromebox, and had both Debian Sid (last updated at around the first quarter of 2019) and windows 10 installed. Booted up, everything working fine.

First thing I did, was to update the flashed image using the firmware utility script (got it up to 2019-12-01) and rebooted.
I immediately noticed that the touchpad stopped working in Debian.
I then tried by updating Debian Sid to the latest packages and kernel, but that changed nothing.
I then rebooted into Windows 10 and surprise, the touchpad works just fine there.

Back to Linux:

- `i2cdetect -l` gives absolutely no result
- Tried to manually load `cyapatp` module (it's not loaded automatically) without results.
- `dmesg` prints nothing about the touchpad.
- Any ideas or other things I could check?

Have a nice day,

MrChromebox, 2020-03-03:
someone else reported this as well, but I don't have a device to troubleshoot here. Nothing changed on LUMPY's trackpad code, and LUMPY and PARROT use the same trackpad (and PARROT doesn't have any issue w/Linux) so this will likely require a bisect with a dozen or so test builds to try and pinpoint the root cause

hkzlab, 2020-03-03:
Any chance this could be related? While the commit is old i see it's in the 2019.12.01 tag:
https://github.com/MrChromebox/coreboot/commit/a378b96ad724d8db356dd0e4079353b7df52db03

Found here:
https://github.com/MrChromebox/coreboot/commits/2019.12.01?after=c32f6317315106319155dff0fee0e83cb4943714+34

MrChromebox, 2020-03-03:
as you can see, that commit is from 2017 and has been in the tree since then. It's unchanged from the 2019.01.04 release, which is functional.

hkzlab, 2020-03-03:
Suspected as much, worth a try!
Any way to download the old release and flash it back? I always used your script, but I don't think i've seen a pathway for a downgrade.

MrChromebox, 2020-03-03:
well if you don't mind flashing some test builds, would love to have your help in debugging. shoot me an email at mrchromebox at gmail if you're interested

hkzlab, 2020-03-03:
I sure am! The laptop is not too useful in the current state, will contact you soon

paranoidandy, 2020-03-16:
Having the same problem here. Have been previously using Cloudready (but with no linux container or shell access). Finally got wind that a firmware update could bring things into the present, but now the touchpad isn't recognised on any OS I try to boot or install. Any updates? I have a soft spot for keeping this machine going. With 6GB of ram it's surprisingly useful!

MrChromebox, 2020-03-16:
@paranoidandy I've found/fixed the issue, just haven't been able to push an update as I've been trying to prep a larger update for all boards

paranoidandy, 2020-03-16:
Awesome news. Happy to help test if needed. Otherwise I'll stop looking for a fix and wait until your update gets pushed. Thanks.

MrChromebox, 2020-03-18:
fixed in 2020.03.17 release!

closed 2020-03-18 by MrChromebox

paranoidandy, 2020-03-18:
Installed and working. Thanks!