MrChromebox/firmware #194: Lenovo N20P black screen after full UEFI firmware install

rom-burner, 2020-03-07:
Exact same issue on CLAPPER chromebook as described in #90. Black screen with backlight on.

Old issue is closed but it's not clear how it was resolved.  Any ideas for non-SPI recovery?

MrChromebox, 2020-03-07:
seems a few Baytrail devices are having issues with the internal display, boot with HDMI attached and should allow you to install the OS, at which point it won't be needed anymore. I'll have this addressed in a update tomorrow hopefully

rom-burner, 2020-03-07:
cool, i'm waiting on a miniHDMI cable - would be happy to test any fixes when up and running if you can't get your hands on any baytrail hardware.  thks!

closed 2020-03-07 by rom-burner

MrChromebox, 2020-03-08:
I have a CLAPPER here, the current firmware works fine on mine. But at least one other person has had the internal display not work for them with the current video init method, so must be some differences in panels used across the line