MrChromebox/firmware #195: Pixelbook 2013 with i5 3427u , the CPU's max clock is 1.8 G hz,  the latest full rom  2019-12-01

whmzsu, 2020-03-09:
Pixelbook 2013 with i5 3427u , the CPU's max clock is 1.8 G hz,  the latest full rom  2019-12-01

From the intela ark 
Processor Base Frequency 1.80 GHz
Max Turbo Frequency 2.80 GHz
Both under Ubuntu 19.10   or Windows 1909. 

Is it normal ???No Turbo Function ??

MrChromebox, 2020-03-28:
am able to reproduce here, am looking into it

terrancewong, 2020-09-29:
Having same issue for years. sometimes CPU goes into turbo frequencies when playing video...

I tried hacking MSR 
```
# wrmsr --all  0x610 0x4c80c800FC80C8h
```

noluck

MrChromebox, 2020-09-29:
yeah I'm not sure, I've had a few other ppl look at it too.

Shercoolcool, 2020-11-27:
same issue with full rom 2020.06.06

bpvarsity, 2022-02-24:
Can confirm the issue with my Chromebook Pixel 2013 as well

bpvarsity, 2022-02-26:
I have not been able to find a kernel or distro that will let the cpu turbo. All of them seem to max at 1.8Ghz. Really wish I could find a way to get the turbo to work

terrancewong, 2022-03-01:
but back some time in 2017~2018 I can run 2.4GHz on Ubuntu (16.04 or similar) XD

bpvarsity, 2022-03-01:
> but back some time in 2017~2018 I can run 2.4GHz on Ubuntu (16.04 or similar) XD

Yeah I haven't gone back that far in versions but I can try it just to see if it can turbo

closed 2026-11-04 by MrChromebox