MrChromebox/firmware #196: ASUS Chromebox 3 (CN65) does not boot without monitor (headless)

lanrat, 2020-03-13:
I have the UEFI Firmware firmware installed on a ASUS Chromebox 3 (CN65) TEEMO device which I plan to use for a small home server. However it does not boot unless I plug a monitor into it.

It looks like headless mode should be working for the UEFI firmware. Do I need to enable it somewhere?

Thanks.

MrChromebox, 2020-03-27:
I have another user with a CN65 running the 2020/03/17 release, and has no issues running it headless (using it as a Daphile server). What firmware version are you running? Also possible the bootloader is the problem

lanrat, 2020-03-28:
I was running `Full ROM / UEFI (MrChromebox-4.11.1 12/01/2019)`. I just booted it again with a monitor and it looks like it may have been a bootloader option. The UEFI boot entries got mixed up.

I just updated to `Full ROM / UEFI (MrChromebox-4.11.2 03/17/2020)` and rebooted headless without issue.

lanrat, 2021-05-06:
I'm having this issue again. The Teemo refuses to boot without a monitor attached. All the boot entries are correct. Running the latest firmware as of today.

If I use one of these [fake monitor plugs](https://smile.amazon.com/Headless-Display-Emulator-Headless-1920x1080-Generation/dp/B06XT1Z9TF) it works great, otherwise if I reboot the device over SSH it will never come back. If I plug a monitor in and do the same there is no issue.

lanrat, 2021-05-06:
Also, when the Teemo gets into this mode where it will not boot. Sometime it takes a while (10-30minutes) for it to work again, even if there is a monitor attached. 

I'm guessing that when it boots without a monitor, some setting gets saved that puts in in an unbootable state and it takes a while and many reboots to get it cleared.

lanrat, 2021-07-26:
Still having this issue, even with the latest firmware.

MrChromebox, 2021-07-26:
not something I've had time to investigate unfortunately. I'd just get a $10 HDMI emulator plug if headless boot is needed

lanrat, 2021-07-26:
I've been running my CN65 with the fake HDMI plug, and it boots most of the time with it, but sometimes it still fails, and sometimes it still fails to boot even with a real monitor attached. When it gets in this state the most reliable way I've found to make it boot is to unplug it and let it sit off for 10 minutes and try again.

I'm starting to think it might be unrelated to this firmware and may be a hardware problem, but I'm not sure how to diagnose further.

Is there a UART somewhere on the board I could attach to to see the boot logs when it gets in this state?

MrChromebox, 2021-07-26:
the CCD interface using a Suzy-Q cable provides UARTs for the CR50, CPU, and EC (ttyUSB0/1/2 respectively), but there's only output on the CPU UART when running a built with serial output enabled, because enabling it by default does slow down boot

lanrat, 2021-07-26:
I can try to get/build a Suzy-Q cable if it would help.

Would it be possible to create a UART enabled firmware image?

MrChromebox, 2021-07-26:
of course, just email me MrChromebox at gmail

lanrat, 2021-07-27:
I'll work on getting the Suzy-Q cable cable and report back.

In the meantime I did a lot more testing today. My CN60 has 2x 2GB ram sticks. I found that when it does boot, sometimes the OS (Debian) sees 2GB of ram, and sometimes it sees 4GB. Even without a full poweroff using the linux `reboot` command can result in a different amount of system memory being available.

This is very strange, so I swapped the internal SSD with another CN65 I have and it showed the exact same behavior of on some boots having 2GB and some 4GB or ram, on an entirely different system with different ram.

MrChromebox, 2021-07-27:
please update to the latest 4.14 firmware on both boxes and let me know if the issue persists

lanrat, 2021-07-27:
I upgraded both boxes to 4.14 before testing.

MrChromebox, 2021-07-27:
<shrugs> not something I can reproduce here on either the CN60 or CN65

CyJaySong, 2024-04-29:
I also encountered this problem

CyJaySong, 2024-05-06:
@lanrat [monodeset](https://linuxmint-installation-guide.readthedocs.io/en/latest/boot_options.html#nomodeset-boot-option)

blackjid, 2024-06-14:
I been having the same issue.

Also, when I'm able to boot with the display plugged in, the computer stop responding as soon as I pull the HDMI plug out. I'm not really sure if the computer reboot or just crash. But I cannot get ssh anymore, and if I plug the display again, it never get signal again..

Any idea?

CyJaySong, 2024-06-14:
@blackjid https://linuxmint-installation-guide.readthedocs.io/en/latest/boot_options.html#nomodeset-boot-option

blackjid, 2024-06-14:
I'm trying to run [Talos Linux](https://www.talos.dev/), I tried adding the `nomodeset`, in the config. But nothing happens, it still reboots when I unplug the disaplay, and doesn't boot after that.

:(

CyJaySong, 2024-06-15:
@blackjid You can try Mint Linux, that method works for me

blackjid, 2024-06-17:
Thanks...,  I tried with a hdmi emulator stick, and it works fine..