MrChromebox/firmware #197: Parrot Keyboard

piggz, 2020-04-03:
Hi, not sure if this is an issue with your FW or not.  I have it installed to enabled me to install arnold the bat chromiumos to my acer c710.  Everything was working fine up until Release 80.  In R80, the keyboard on the laptop doesnt work once the os is installed to the hard disk (works fine running from usb).

An external usb keybord works fine.  I wondred if that was any possibility that the fw could be messing things up some how...i know its a stretch!

See also https://github.com/arnoldthebat/chromiumos/issues/306

MrChromebox, 2020-04-03:
If you're still able to enter the firmware boot menu, I highly doubt it's a firmware issue, but worth pulling the boot log from ChromiumOS and see. After a fresh boot, open a shell and run `dmesg > dmesg.txt` and then post the contents of that file on pastebin (eg) and report the link here

piggz, 2020-04-03:
Here you go, https://paste.mozilla.org/nifff0nu 
I agree, its unlikely to be fw, just thought id check.  Has happened between chromos 78 and 80, and only on hd installs

MrChromebox, 2020-04-03:
I see some weirdness going on, but the ps2 keyboard looks fine best I can tell

piggz, 2020-04-03:
I guess im at the mercy of arnold the bat, but he hasnt been as responsive! :D

piggz, 2020-04-04:
Ok, i can close this.  In a separate problem, i was aware that the right hand usb ports didnt work.  I took the laptop apart, couldnt find anything loose, and put it back together.  I booted up and the keyboard is working!

closed 2020-04-04 by piggz

piggz, 2020-04-06:
Finally figured it out, it was a 3rd part power supply causing issues!

MrChromebox, 2020-04-06:
crazy how that was an issue!

piggz, 2020-04-06:
Agreed! But crazier is the combination of R80 + Charger...... the charger was always a bit dodgy in how it effected the touchpad, it would make the touchpad almost unusable, but never stopped the keyboard working.  Then, when i upgraded my daughters R78 to R80, we could get as far as entering the username, but then the keyboard would stop working before entering the password.  I unplugged the charger, and could type again.  I then looked for the good charger, and had no issues with it, with either the touchpad or keyboard.  I probably should look for the other good charger and throw this one out!