MrChromebox/firmware #198: Getting Tianocore some .efi ROMs without using disk

Hurricos, 2020-04-11:
Hi there,

Can you confirm whether it is possible to build Tianocore in such a way that it can see .efi programs added in any way to the flash? (Can `cbfstool` be used to add these ROMs?)

I'd like to build (among other things) a minimal iPXE .efi that's visible to Tianocore and which can reside on flash. The usecase here is that regardless of what's on the disk, I can use a combination of Tianocore and iPXE to handle block device booting, but also set up NICs and pull down memtest.efi and so on.

MrChromebox, 2020-04-11:
Tianocore has no concept or awareness of the CBFS, so simply adding files that way won't help. You need to add them to Tiano's FV (firmware volume) but I'm not sure if simply dropping an .efi driver in there is enough - believe it needs to be registered as well

Hurricos, 2020-04-11:
So that's what FV means. I'm somewhat familiar with the tools that come with EDK2. Are you suggesting tinkering with tools like `tianocore/BaseTools/BinWrappers/PosixLike/*Fv*` to modify the `UEFIPAYLOAD.fd` (or one of its precursors or derivatives)?

MrChromebox, 2020-04-11:
I've not familiar with the specific tools used in the build process, but you can see how I add a binary driver to my build here:  
https://github.com/MrChromebox/edk2/commit/daa1f83b55cc45cbcf512d5eeb8398521568d860

and if i wanted to manipulate a prebuilt payload after the fact, I'd use:  
https://github.com/LongSoft/UEFITool

closed 2026-11-04 by MrChromebox