MrChromebox/firmware #199: (Only partially related to firmware) Best location to document experience?

Hurricos, 2020-04-11:
Hi there,

You previously helped me previously to rebuild Tianocore from scratch to replace the Tianocore bootsplash. I'd like to make sure other people don't trip over the same hurdles as I did by documenting my process. Do you have a suggested location for me to start documenting what I did to get things to build properly?

https://doc.coreboot.org/ does not seem like the place for documenting this type of experience; the wiki at https://www.coreboot.org/Payloads is being (eventually) retired. 

Seems like https://wiki.mrchromebox.tech/ could be a good place to start. Can you confirm?

If you feel it's inappropriate to build guides for this type of thing then feel free to close this issue. I see a lot of red links [here](https://wiki.mrchromebox.tech/) and would contribute to try and fill in whatever you think I should if you think that's a worthy place to do so.

MrChromebox, 2020-04-11:
I think having a wiki page for building from my repo is a definitely good idea. Right now, I've not set up the wiki as anything more than just a few standalone pages, with the framework to mimic the existing site layout (of which I'm unsure is the best way to organize things for a wiki, and why I haven't fleshed out those pages)

Hurricos, 2020-04-14:
Hey there,

Could you help me create an account on your wiki so that I can create such a page while I still have the process fresh in my mind?

Thanks.

MrChromebox, 2020-04-14:
I'll need to figure out how to add accounts or allow registration, I had someone else set it up awhile back. or if you just want to email me in text format I can do all the wiki bits

Hurricos, 2020-04-14:
I'll send 'em to you in .md format.

dragon788, 2020-08-19:
@MrChromebox have you considered enabling the Github wiki here and using a group to manage wiki editors? It might be easier to manage than MediaWiki. Otherwise you could maybe use something like a firmware.mrchromebox.tech subdomain and publish a Github Pages there to allow easier contributions through Github.

MrChromebox, 2020-08-19:
> @MrChromebox have you considered enabling the Github wiki here and using a group to manage wiki editors?

I'm not seeing that option -- only to restrict editing to collaborators, or have it open to everyone

closed 2026-11-04 by MrChromebox