MrChromebox/firmware #200: No fan on pixelbook+ubuntu desktop 18.04

williamyoon, 2020-04-24:
Hi, I have flashed mr chomebox bios and installed ubuntu 18.04 on my google chromebook pixel 2013 (non lte).  I noticed the fans are not running.

Is this caused by the firmware?

MrChromebox, 2020-04-24:
fan is set to auto mode by the firmware on boot. mine fan speed is 1200 rpm, which is inaudible. Run something that stresses the CPU (like `stress -c 4`) and you will hear the fan kick in.

unrelated, it's not ideal to be running a 2 year old version of Ubuntu, even if it is LTS

LPFchan, 2021-09-18:
Hey, just chiming in on an old topic.
There seems to be something going on with the Linux kernel. 
Anything from 5.3 and onwards will kill the fans to almost standstill while running anything 5.2.x or prior will ramp up the fan as necessary. (I'm currently using mainline 5.1.21 on Ubuntu 21.04)
I'm a bit confused as to EC is supposed to be handling all the fan controls? (#92)

Also I'm okay with the fans running a bit louder than it currently is and wish to run the device a bit cooler, but I couldn't access or tune PWM with lm_sensors or fancontrol because... EC. 
Is there any way I could ramp the fan speed up a bit more aggressively?

MrChromebox, 2021-09-18:
Google's ectool, which is part of the ChromeEC source, is the only user tool for interfacing with the fan, and getting it to compile/run is increasingly difficult. I have the fan set to auto mode on boot so the kernel shouldn't be screwing with it. Give me a distro I can live/boot and test which exhibits the problem and I'll see what I can find

LPFchan, 2021-09-19:
@MrChromebox Distro is **Ubuntu 21.04**. It seems like there's two different bugs currently affecting **_link_** and I'm not sure if those two are independent or correlated to each other. I've spent some time testing out many versions of mainline(upstream) kernels and here's what I found out:

Here are the problematic behaviours: Fan halts itself every 3.9 seconds approximately and doesn't speed up under load. And system doesn't go to suspend and wakes up instantly. `cat /var/log/syslog` reports `Device cros-ec-lightbar.6.auto failed to suspend: error -22`. Same thing happens both on the Ubuntu kernel and the Mainline(Upstream) kernels so I doubt this is distro-specific.

- Ubuntu 21.04 Stock kernel (`5.11.0-16-generic`, `5.11.0-34-generic`): Slow fan, Failed suspend
- Upstream `5.11.0-051100-generic`: Same behaviour
- Upstream `5.10.0-051000-generic`: Same behaviour
- ...
- Upstream `5.10.0-051000rc1-generic`: Same behaviour (Slow fan, Failed suspend)
- Upstream `5.9.16-050916-generic`: Suspend works, Fan is slow on boot. However when I wake up from suspend the fan ramps up as normal.
- ...
- Upstream `5.8.0-050800-generic`: Suspend works, Fan works sometimes and sometimes don't.
- Upstream `5.7.19-050719-generic`: Suspend and Fan works as it should pretty reliably

I can definitely work my way around the bug by setting 5.7.19 as grub default and daily driving it, but it would be rewarding to find out where the bug came from and see it fixed. Though I'm pretty sure this is a kernel issue but I just wanted make sure it's not UEFI/BIOS thing before taking this to bugzilla.kernel.org

MrChromebox, 2021-09-19:
what happens if you blacklist the lightbar or chrome-ec-lpc driver(s)?

LPFchan, 2021-09-20:
Blacklisted the `cros_ec_lightbar` module only at first (on the latest 5.11.0-34-generic kernel) 
Fan is slow on boot, Suspend works, and then when it wakes up from suspend the fan starts to ramp up as normal. It behaves like 5.9.16-050916-generic. 

Then I blacklisted the `cros_ec_lpcs` module, now **fan and suspend works as it should**. Though it seems like every cros module other than `cros_kbd_led_backlight` is gone after that? 
```
$ lsmod | grep cros
cros_kbd_led_backlight    16384  0
```

(I checked the dependencies but I don't really understand why other modules went away with it)
```
$ modinfo -F depends cros_ec_lpcs
cros_ec
$ modinfo -F depends cros_ec     

```

MrChromebox, 2021-09-20:
sounds like something broke in those drivers and/or the way they interact with the rest of the kernel in 5.9+

LPFchan, 2021-09-20:
Do you know for what purpose are those drivers there for in the first
place? I've been running it blacklisted for a while and it seems like
everything runs just as normal.

erkserkserks, 2021-12-06:
I also had this fan issue with my Chromebook Pixel 2013 running Ubuntu 20.04 with kernel 5.11. I solved it by installing an older 5.4 kernel.

I followed this approach to keep the kernel on an older version: https://askubuntu.com/a/1310033
Since I was already on a new kernel, my steps were as follows:
```
# This will pull in the latest 5.4 Ubuntu kernel, and receive subsequent 5.4.0-x patches too
sudo apt install linux-generic-hwe-18.04

# Identify which kernel packages are going to be removed
dpkg -l | grep -o -P "^ii\s+linux-[\S]*-5.(8|11|13).0[\d\S]+"

# Remove newer kernels
sudo apt remove -y $(dpkg -l | grep -o -P "linux-[\S]*-5.(8|11|13).0[\d\S]+")

# This prevents installation of, or further upgrades of the package that results in a newer kernel being installed
sudo apt-mark hold linux-generic-hwe-20.04
```

erkserkserks, 2021-12-06:
I also had this fan issue with my Chromebook Pixel 2013 running Ubuntu 20.04 with kernel 5.11. I solved it by installing an older 5.4 kernel.

I followed this approach to keep the kernel on 5.4.x: https://askubuntu.com/a/1306420

robotcarnival, 2022-02-23:
LPFchan and MrChromebox - you guys are brilliant - blacklisted cros_ec_lpcs by following this procedure (https://linuxconfig.org/how-to-blacklist-a-module-on-ubuntu-debian-linux) and worked like a charm!!! Thanks again!!!!

bpvarsity, 2022-02-23:
> LPFchan and MrChromebox - you guys are brilliant - blacklisted cros_ec_lpcs by following this procedure (https://linuxconfig.org/how-to-blacklist-a-module-on-ubuntu-debian-linux) and worked like a charm!!! Thanks again!!!!

This worked! Just tested it out on Elementary 6.1 and fan works perfectly now. Thank you thank you

navyfighter12, 2022-09-19:
I am having an issue with the fans with Manjaro. I have tried 6 different kernels and no change there. 

Its not like when on Mint, Elementary, Ubuntu etc where the fan doesn't spin or barely spins instead it seems to be having a fan curve issue. I can run stress -c 4 and it doesn't ramp up until its 88-92 C and when it does its at 50% max fan speed. I have even gotten the temps to 94 C and still no more than 50%. The problem is at 50% it can only maintain the 90-92 C temp not lower it. If I manually run ectool fanduty 100 I can kick those fans up and cool the cpu down quickly. 

Is there anyway to change the fan curve? Lately I have been booting up and manually setting fans to 50% and when I do more intense actions I bump it up to 70-80% depending on the temp. I would prefer this to be automatic and have the ability to kick those fans beyond 50%. 

Thanks for the help.

navyfighter12, 2022-09-19:
I just found this for changing the fan temp points 

[](https://community.frame.work/t/changing-the-fan-temperture-points-with-ectool/17355)

however I am getting some different results than the guide shows. Not sure why the fan off is 1 and fan max is 0 vs having temps like the example. Maybe just a different version but thought it was odd that there was not a fan max temp.

ectool thermalget 
sensor  warn  high  halt   fan_off fan_max   name
  0      368   374    377      1       0     PECI
  1        0   361    371      1       0     ECInternal
  2        0     0      0      1       0     I2C-Charger-Die
  3        0     0      0      1       0     I2C-Charger-Object
  4        0     0      0      1       0     I2C-CPU-Die
  5        0     0      0      1       0     I2C-CPU-Object
  6        0     0      0      1       0     I2C-Left C-Die
  7        0     0      0      1       0     I2C-Left C-Object
  8        0     0      0      1       0     I2C-Right C-Die
  9        0     0      0      1       0     I2C-Right C-Object
 10        0     0      0      1       0     I2C-Right D-Die
 11        0     0      0      1       0     I2C-Right D-Object
 12        0     0      0      1       0     I2C-Left D-Die
 13        0     0      0      1       0     I2C-Left D-Object
 14        0     0      0      1       0     Battery

navyfighter12, 2022-09-19:
Okay well I found a solution. This python service lets you adjust the curve and its amazing! Starts on boot and I can set it to a preset curve profile or make my own. 

Here it is for those who want this control as well 

[](https://github.com/TamtamHero/fw-fanctrl)

MrChromebox, 2022-09-19:
LINK has an issue with the CPU speed being pegged at the base clock (1.8GHz) and not scaling up or down. That directly affects the CPU temp and therefor fan duty cycle.  Keep an eye out on temps if you're going to be adjusting the speed manually

navyfighter12, 2022-09-19:
> LINK has an issue with the CPU speed being pegged at the base clock (1.8GHz) and not scaling up or down. That directly affects the CPU temp and therefor fan duty cycle. Keep an eye out on temps if you're going to be adjusting the speed manually

Do you know of a better way to do this? Its very annoying that the fans max out at 50% when set to auto. I am not a fan of the cpu holding at 90C when it could cool itself down if it just increased the fan speed

MrChromebox, 2022-09-19:
I don't, just letting you know what the root cause is

navyfighter12, 2022-09-19:
> I don't, just letting you know what the root cause is

Ah I made a mistake in posting in this issue. I have the Chromebook Pixel 2015 LS Samus. I believe the 1.8GHz just affected the 2013 version unless I am mistaken.

MrChromebox, 2022-09-19:
that's correct, the 2015 version is SAMUS

user8446, 2022-12-02:
I'm on a Dell Chromebox 3010 and I added `blacklist cros_ec_lpcs` to `/etc/modprobe.d/blacklist.conf` per the above notes.

It did remove the `cros_ec_lpcs EC ID not detected` error on boot but it appears the fan still just stays at one speed. I installed `lm-sensors` and it can see temps but not fan speed. Does the Chromebox not even have fan speed control?

MrChromebox, 2022-12-02:
@user8446 please open a separate issue for your device, it does not have an embedded controller