MrChromebox/firmware #201: UEFI full firmware - unable to get to shell sub menu

enchekamat, 2020-04-24:
I'm on Chromebook Pixel 2013 (LINK) and have flashed latest full UEFI firmware using MrChromebox firmware utility. All went well and I've installed Linux (Clear) and able to access shell menu and all. Recently I installed another distro (Fedora) but now I'm unable to access the sub menu of shell ( 2nd and 3rd menu - One time boot menu and and boot menu configuration) to change boot order. Whenever I press Enter on these menu I only got a blank screen (backlight was on though) but no menu doesn't matter what key I press. The laptop will power off when I press Power button. So now i'm only stuck to booting to Clear Linux and nothing else. I tried to run the firmware utility again but got an error message telling me that ' Unable to read current firmware; cannot continue. '.

MrChromebox, 2020-04-24:
> I tried to run the firmware utility again but got an error message telling me that ' Unable to read current firmware; cannot continue. '.

and after that, it tells you exactly how to fix the problem! You need to add the required kernel parameter so flashrom can read/write the firmware. Then you can use my script to clear the NVRAM and you'll be good to go.

closed 2020-04-24 by MrChromebox

enchekamat, 2020-04-24:
Thanks but pretty much a newbie with Linux. Is it in GRUB that i need to add this? cat: /etc/default/grub: No such file or directory

MrChromebox, 2020-04-24:
I don't know how it would be done on Clear Linux offhand, you'd need to check its documentation

enchekamat, 2020-04-25:
Thanks! Finally able to access the shell menu again after configuring the systemd boot loader.