MrChromebox/firmware #204: USB 3 hard drive enclosure with SSD not detected on reboot

nschimme, 2020-05-07:
I noticed that with my Asus CN60 that it is unable to detect USB 3 hard drive enclosures on reboot. In order to detect the hard drive enclosure I need to physically unplug it and plug it back in for it to be detected by the Ubuntu 20.04.

The [hard drive enclosure](https://www.amazon.com/gp/product/B01M08LCXW) is powered by the JMS578 chipset and I've flashed it to run the [latest firmware](https://wiki.odroid.com/odroid-xu4/software/jms578_fw_update). This problem does not occur on any other PCs which leads me to believe it is a UEFI BIOS issue tied to the Chromeboxes.

MrChromebox, 2020-05-07:
the first-gen Chromeboxes have somewhat flaky USB ports, especially the front two. Which port(s) have you tried using? IME, the rear ones work better for external storage devices.

nschimme, 2020-05-07:
No success on either front and rear ports

nschimme, 2020-05-07:
```
$ dmesg | grep -i usb
[    0.173349] ACPI: bus type USB registered
[    0.173349] usbcore: registered new interface driver usbfs
[    0.173349] usbcore: registered new interface driver hub
[    0.173349] usbcore: registered new device driver usb
[    0.668528] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    0.668570] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    0.668596] uhci_hcd: USB Universal Host Controller Interface driver
[    0.668803] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 1
[    0.670034] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.04
[    0.670039] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.670043] usb usb1: Product: xHCI Host Controller
[    0.670046] usb usb1: Manufacturer: Linux 5.4.0-29-generic xhci-hcd
[    0.670049] usb usb1: SerialNumber: 0000:00:14.0
[    0.670215] hub 1-0:1.0: USB hub found
[    0.670967] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 2
[    0.670972] xhci_hcd 0000:00:14.0: Host supports USB 3.0 SuperSpeed
[    0.671011] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.04
[    0.671016] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.671019] usb usb2: Product: xHCI Host Controller
[    0.671022] usb usb2: Manufacturer: Linux 5.4.0-29-generic xhci-hcd
[    0.671025] usb usb2: SerialNumber: 0000:00:14.0
[    0.671139] hub 2-0:1.0: USB hub found
[    1.828997] usb 1-2: new high-speed USB device number 2 using xhci_hcd
[    1.849745] usb 1-2: New USB device found, idVendor=11b0, idProduct=6298, bcdDevice= 1.08
[    1.849758] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.849765] usb 1-2: Product: SNA-DC/U
[    1.849770] usb 1-2: Manufacturer: Kingston
[    1.849775] usb 1-2: SerialNumber: 38313132464C594E53202020
[    1.976984] usb 1-4: new full-speed USB device number 3 using xhci_hcd
[    2.126419] usb 1-4: string descriptor 0 read error: -22
[    2.126429] usb 1-4: New USB device found, idVendor=04ca, idProduct=3007, bcdDevice= 0.01
[    2.126433] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.253050] usb 1-6: new full-speed USB device number 4 using xhci_hcd
[    2.403587] usb 1-6: New USB device found, idVendor=04f2, idProduct=1338, bcdDevice= 1.02
[    2.403607] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.403610] usb 1-6: Product: USB Wireless HID Receiver
[    2.403613] usb 1-6: Manufacturer: Chicony
[    2.409642] usb-storage 1-2:1.0: USB Mass Storage device detected
[    2.409807] scsi host4: usb-storage 1-2:1.0
[    2.409987] usbcore: registered new interface driver usb-storage
[    2.411621] usbcore: registered new interface driver uas
[    2.426977] usbcore: registered new interface driver usbhid
[    2.426983] usbhid: USB HID core driver
[    2.432110] input: Chicony USB Wireless HID Receiver as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:04F2:1338.0001/input/input2
[    2.489222] hid-generic 0003:04F2:1338.0001: input,hidraw0: USB HID v1.11 Keyboard [Chicony USB Wireless HID Receiver] on usb-0000:00:14.0-6/input0
[    2.489529] input: Chicony USB Wireless HID Receiver Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:04F2:1338.0002/input/input3
[    2.549212] input: Chicony USB Wireless HID Receiver System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:04F2:1338.0002/input/input4
[    2.549310] input: Chicony USB Wireless HID Receiver as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:04F2:1338.0002/input/input5
[    2.549450] hid-generic 0003:04F2:1338.0002: input,hiddev0,hidraw1: USB HID v1.11 Device [Chicony USB Wireless HID Receiver] on usb-0000:00:14.0-6/input1
[    2.549635] input: Chicony USB Wireless HID Receiver as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.2/0003:04F2:1338.0003/input/input6
[    2.549749] hid-generic 0003:04F2:1338.0003: input,hidraw2: USB HID v1.11 Mouse [Chicony USB Wireless HID Receiver] on usb-0000:00:14.0-6/input2
[    2.549891] hid-generic 0003:04F2:1338.0004: hiddev1,hidraw3: USB HID v1.11 Device [Chicony USB Wireless HID Receiver] on usb-0000:00:14.0-6/input3
[   10.517279] usbcore: registered new interface driver btusb
[   10.603599] usb 1-4: USB disconnect, device number 3
[   10.604334] usbcore: registered new interface driver ath3k
[   10.908994] usb 1-4: new full-speed USB device number 5 using xhci_hcd
[   11.058272] usb 1-4: string descriptor 0 read error: -22
[   11.058280] usb 1-4: New USB device found, idVendor=04ca, idProduct=3007, bcdDevice= 0.02
[   11.058283] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
```

Then after I unplug the enclosure and plug it back in:
```
$ dmesg | grep -i usb | tail -n 7
[   11.058283] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  157.941056] usb 2-3: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[  157.962229] usb 2-3: New USB device found, idVendor=152d, idProduct=0578, bcdDevice=31.02
[  157.962233] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  157.962235] usb 2-3: Product: USB to ATA/ATAPI Bridge
[  157.962237] usb 2-3: Manufacturer: JMicron
[  157.962239] usb 2-3: SerialNumber: 0123456789ABCDEF
```

MrChromebox, 2020-05-07:
I'm not sure if it's a hardware issue or a coreboot issue, but it's not a Tianocore/UEFI issue.

I do have a new test build which reworks the boot device detection and may work better, shoot me an email if you'd like to test -- mrchromebox at gmail

moriel5, 2020-08-04:
It's probably a hardware/firmware issue with the SATA to USB case.

I'm having similar issues with mine (also Orico, also a JMicron chip, however it is USB-C with support (USB-wise, SATA3 will never reach it) for 10Gbt/s), and the issues persist with other PCs as well (am able to run an old SATA2 Momentus 250GB HDD (0.45A), which sometimes unmounts, but not a 1TB SATA3 SpinPoint HDD (0.85A), while Seagate's 1TB Barracuda (1A) works just as well as the aforementioned 250GB HDD).

closed 2026-11-04 by MrChromebox