MrChromebox/firmware #206: Firmware Request

jwatson123, 2020-05-16:
Hi,
I've got an HP 11 G6 (SNAPPY) running Intel Apollo Lake, I love the RW_LEGACY firmware it's got running but I was wondering if there are any plans for a full UEFI firmware for Apollo Lake in the future?
Am happy to trail any test builds if there are any in the works

Thank you and keep up the good work!

MrChromebox, 2020-05-16:
no plans for UEFI firmware for Apollolake or Geminilake boards, too many things don't work still and don't want to deal with the increased support load from people trying to run Windows on these devices

closed 2020-05-16 by MrChromebox

rolandE09, 2022-01-16:
> no plans for UEFI firmware for Apollolake or Geminilake boards, too many things don't work still and don't want to deal with the increased support load from people trying to run Windows on these devices

but is it possible to install windows on it? or not at all?

MrChromebox, 2022-01-16:
> but is it possible to install windows on it? or not at all?

Absolutely not