MrChromebox/firmware #207: Boot From USB .iso Not functioning

flub9, 2020-05-18:
When using the UEFI BIOS, when I attempt to boot an iso from a usb via the Boot Menu, I am being returned to the main BIOS menu, with seemingly no attempt to boot. I tested with both of my devices USB ports, and it is definitely not a USB port acting up.

MrChromebox, 2020-05-18:
this is clearly an issue with your USB stick, the ISO, or how you are writing it to the USB. There is no firmware-level issue booting UEFI compatible USB media

flub9, 2020-05-18:
I'll try via a USB, also, when I flashed the BIOS, there was a preexisting Gallium OS instance on the drive, could that have affected it?

MrChromebox, 2020-05-18:
I don't know what driver you're referring to. The contents of the internal storage are irrelevant, are are the contents of the USB drive if you write the ISO to it properly (obv, you can't just copy the ISO to USB).

If you're trying to boot GalliumOS via USB to install it, then ensure the ISO is written to USB properly using one of the approved methods (dd, Etcher, or Rufus in DD mode).

flub9, 2020-05-18:
My bad, by drive I meant the laptop's drive. I'm working on getting the ISO on the SD right now. I'll report back with how it worked out.

flub9, 2020-05-18:
SD works, cause of the issue seems to have been my USB. Thanks for the assist!

closed 2020-05-18 by flub9