MrChromebox/firmware #208: Swanky screen different than previous releases

jacobopantoja, 2020-05-24:
I updated to the latest 2020.03.17 in my Toshiba Swanky, and I've noticed a few differences:
* Coreboot is now low resolution (I would say 640x480), and full screen. Previously, it always booted at native resolution (1920x1080). Perhaps it was set by me to 1920x1080, but see next point.
* Coreboot menus are gone. I cannot longer set, e.g., the boot resolution.
* Backlight brightness values are different. When setting low values from my ArchLinux, e..g 100 or even 1 which is the minimum I can set, the perceived brightness is higher than with previous release. This is problematic when using the laptop in total darkness.

MrChromebox, 2020-05-24:
The video init method was changed for the current release, resulting in a fixed boot resolution of 1024x768, and likely the change in backlight controls. It will be reverted for the upcoming release as it created more problems than it fixed. 

There was never an ability to change the resolution via the firmware menus.

jacobopantoja, 2020-05-24:
> There was never an ability to change the resolution via the firmware menus.

Thanks for the answer, that was super-fast. Perhaps I am mixing memory images from other computers. I was not careful to save past firmware binaries so I could compare. Nevertheless it is ok, the main problem I could see is the brightness problem.
If I can help in any way, please tell me

MrChromebox, 2020-05-24:
you'll just need to wait for the next release, which hopefully will be in the next few days

MrChromebox, 2020-06-07:
fixed in 4.12 release

closed 2020-06-07 by MrChromebox