MrChromebox/firmware #209: corrupt download for lars chromebook

jefferysboyd, 2020-05-25:
hi mr chromebox,
when i use the firmware utility script to install full rom, it fails with error message - corrupt rom can't flash.
it's a c771t acer chromebook lars. i ran the script from both chromeos and archlinux and got the same result.

MrChromebox, 2020-05-25:
please provide the exact error, a pic would be best

jefferysboyd, 2020-05-25:
Can't upload a photo but it reads...

Downloading Full ROM firmware
(coreboot_tiano-lars-mrchromebox-20200317.rom)

Firrmware download checksum fail; download corrupted. cannot flash.

Press [Enter] to return to the main menu.

MrChromebox, 2020-05-25:
the file is present and checksum correct, so I can only assume that the download is being corrupted in transit somehow and the script is correctly preventing you from flashing a bad file.

if you check your /tmp dir, does the file exist? is it 16MB? what's the sha1sum of the file?

jefferysboyd, 2020-05-25:
yes, no, 3ce5e758982eb97da2637126a42be0f803b7082

MrChromebox, 2020-05-25:
if it's not 16MB, then it's getting truncated on download, hence the incorrect checksum. Doesn't appear to be a server-side issue

jefferysboyd, 2020-05-25:
ok thanks. i will defenestrate my laptop on my own from here.
hey wait a minute. 16 MB is pretty small, no? Can my internet service really be that terrible?

MrChromebox, 2020-05-25:
hard to say, I'm just saying I downloaded and verified it myself, so it's not a universal problem

jefferysboyd, 2020-05-25:
coreboot updated in May. is the script downloading something it doesn't expect?

MrChromebox, 2020-05-25:
no, that's the correct filename currently. I've not yet released an update based on coreboot 4.12

jefferysboyd, 2020-05-25:
ok thanks for your help

closed 2020-05-25 by jefferysboyd