MrChromebox/firmware #212: Helios not supported?  Chromebook is listed as supported.

John-IT, 2020-06-15:
Utility script indicates WP disabled but only options 3 and 4 are available.  Choose 1 and result is Unknown or unsupported device (helios); cannot update RW_LEGACY firmware.
This is a 
**   Device: ASUS Chromebook Flip C436FA (HELIOS-YVRQ)
Which is listed as a supported device.

MrChromebox, 2020-06-15:
https://mrchromebox.tech/#devices clearly lists no firmware options currently available for HELIOS or any of the new Comet Lake devices

John-IT, 2020-06-15:
I read you results incorrectly.  BTW, thank you for your quick response, your service to the community and this awesome project.  May I request this device be supported?  Willing to help in any way as I did not buy a 16G, 512 SSD device to just run chrome. :-)  ATM it is a toy, so we can have our way with it.  Device ready for sacrifice.

MrChromebox, 2020-06-15:
RW_LEGACY/legacy boot mode is broken on the stock firmware side of things; I'm working with some Google folks to try and resolve it but it's slow going and I can't say I anticipate it being resolved in the near future.

UEFI firmware is in testing, and I could whip up something for HELIOS no problem, but the OS/driver support outside of ChromeOS isn't great, though everything worked pretty well under Linux outside of audio when I tested quickly on AKEMI. Shoot me an email if you'd like to try a test build

iggut, 2020-11-27:
Hi! Any chance there was any progress with UEFI on Helios. I also have an C436 that I would love to play around with. Any chance to get a test firmware for UEFI please?

MrChromebox, 2020-11-27:
I plan on adding UEFI firmware for all Cometlake devices with my upcoming coreboot 4.13-based release

infinity0, 2021-05-30:
Thanks for the new release & for this project in general! Are your RW_LEGACY discussions happening anywhere public e.g. https://chromium-review.googlesource.com/ ?

Since some Cometlake devices are already list [RW_LEGACY as supported](https://mrchromebox.tech/#devices), I'm wondering if the lack of a checkbox for the other devices means "definitely not supported" or "support not yet tested or confirmed but there's a good chance it'll work". (I'd also be happy with Full ROM, but Suzy Q is sold out everywhere in the UK currently.)

MrChromebox, 2021-05-30:
Cometlake Chromebooks don't have working RW_LEGACY firmware, only Chromeboxes. There are bugs in the stock firmware preventing Tianocore from booting as a RW_LEGACY payload. I've tried to work around them to no avail

infinity0, 2021-05-30:
Thanks for your quick response - could you share the specific details of these bugs? I wonder if it might be possible to binary monkey-patch the stock firmware to fix them directly, after disabling WP of course. That would allow dual-boot at least, for those of us that prefer it. I could be happy to try to figure this out myself once I get one of these elusive Suzy Qables.

MrChromebox, 2021-05-30:
you can't patch the RW firmware because it will fail verified boot / trigger booting the other RW firmware slot or to recovery mode.

the issues I've run into are numerous and not easily worked around. I've patched Tianocore to hell and back and still can't get it booting

infinity0, 2021-05-31:
> you can't patch the RW firmware because it will fail verified boot / trigger booting the other RW firmware slot or to recovery mode.

Since the patching requires disabling WP, one could also patch away all of these checks in theory if I understand right - e.g. the Full ROM does not even contain these checks. Of course the hard part would be finding out which part of the stock firmware binary code to patch out, I'm just armchair theorising here.

I understand it might be a lot of work, I'm not asking anyone to do it - rather I'm curious about this topic as it's my first time looking into Chromebooks and potentially it might be fun to try to do myself, if given enough details so I have a good idea of a concrete goal to be aiming for.

infinity0, 2021-05-31:
Although the next time the firmware gets upgraded all the patches would get nuked I suppose, so maybe not such a nice thing in practise to be running.

MrChromebox, 2021-08-12:
closing as HELIOS and other CML Chromebooks are now supported