MrChromebox/firmware #213: Asus Chromebox 2 Cn62 (i7) 4k 60hz DP Output

bkacz88, 2020-06-16:
4k 60hz not achievable out of Displayport on Asus chromebox 2 cn62. Max refresh rate is 30hz at 2160p currently. Running latest Win 10 version available. (We've spoke on reddit).

dragon788, 2020-08-19:
Are you using a DisplayPort to HDMI adapter or DisplayPort directly into your monitor/TV? For 4k@60Hz you need a DP 1.4 capable monitor or a DP 1.4 to HDMI 2.0 adapter to get 60Hz.

backcountrymountains, 2021-08-03:
Just looking to necro this because I, too, am interested in 4k60hz out of the displayport. I'm pretty sure you need an i3 or i7 processor. @bkacz88, did you try a MrChromebox firmware with a modified CDCLK?

MrChromebox, 2021-08-03:
the problem is even if the firmware initializes the DP output with the correct cdclk etc, the i915 driver gives no shits, and will basically completely reinitialize the GPU, including the cdclk. I just don't know enough about the i915 driver to be able to recommend a way to force it to use the hardware's capabilities

bkacz88, 2021-08-05:
@backcountrymountains No i didn't try with a modified CDCLK. Im still on 4.11.2.

closed 2026-11-04 by MrChromebox