MrChromebox/firmware #214: Question: Is there a hardware speed limit between USB 3.0 and the ethernet on chromeboxes?

wicadmin, 2020-06-28:
Is there a physical hardware limit around 40 MB/s on chromeboxes? Especially between the USB 3.0 ports and ethernet port?

MrChromebox, 2020-06-28:
a hard limit for what? I've had no problem saturating a gigE link when reading from the SSD (~100MB/s)

wicadmin, 2020-06-28:
I was looking at this review where it mentions 
"Although the Chromebox features four USB 3.0 ports, I measured max sequential write speed at around 42MB/s"

https://www.anandtech.com/show/7850/asus-chromebox-review

It went on to suspect the add but curious if this was measure, especially for an LTE modem plugged into the usb port, how much throughput can go to the ethernet.

MrChromebox, 2020-06-28:
that's a review from 2014 with the stock firmware, it's certainly not indicative of performance with my UEFI firmware currently. I can hook up a USB3 SSD and see what the xfer rates are

wicadmin, 2020-06-29:
That would be great. Thank you.

MrChromebox, 2020-06-29:
just booted up ZAKO (HP CB-1) and did a 4GB ISO transfer between a USB3-attached SSD (which is also the boot drive) and my NAS which are connected by gigE, and the transfer rate in both directions was ~92MB/s.

closed 2020-06-29 by wicadmin