MrChromebox/firmware #216: Caroline S-Pen input no longer working

thandlovsky, 2020-07-11:
I have Windows installed on a Caroline Samsung Chromebook Pro and the stylus was working, but after upgrading the firmware from coreboot 4.11.1 (2019.12.01) to 4.12 (2020.06.06) it no longer works.
( I seemed to have skipped the (2020.03.17) 4.11.2 release, so I am not sure if the S-Pen input was working for that revision.)

It was tested and confirmed to be working in a liveboot of GalliumOS that was used to flash the firmware via the utility, but not afterwards.

Does anyone know if there is a way to safely downgrade the UEFI Firmware to an earlier version, or if the stylus support can be patched somehow?

I have a CH341A USB programmer available if that can be used to help at all.

MrChromebox, 2020-07-11:
I'm not aware of any change to anything input-wise on CAROLINE, or any other Skylake device, but possible something under the hood changed. Problem is none of my Skylake devices have touch/pen input.

thandlovsky, 2020-07-11:
I see, thank you for your help and firmware! 

I do need the pen input more than the other fixes that came with the new firmware though.

Is there a way for me to use/modify the scripts to flash the earlier versions of the firmware? Would this be safe?

If I am able to flash the older version, I might be able to see if there are changes to the recognized devices and see if there is anything I can do.

MrChromebox, 2020-07-11:
shoot me an email at mrchromebox at gmail and I'll get your sorted

waitlamp, 2021-03-11:
Same situation,

Windows 10 20H2 cannot find Touchpad and TouchScreen driver automatically. but [crostouchpad](https://github.com/coolstar/driverinstallers) driver solved. The stylus still not work. Some Unknown devices in device manager. I searched hardware ID, all of them are audio devices, not digitizer. 

I also tried Ubuntu 20.04 Live CD. Touchpad and TouchScreen work well, TPM moudle not work as windows ( not important ), but digitizer stylus still not work, maybe the newer firmware didn't drive Wacom W9013 iic devices properly.

I send an Email to you. Could you please reply me and tell me the way to flash a old version firmware, thanks.

waitlamp, 2021-03-11:
Add: I find a UEFI shell Utility can scan i2c bus.

I scanned i2c bus[0] (empty, not on picture) to I2C bus [4],
It listed 6 address. 4 of I2C bus [4], 2 of I2C bus [1]

Caroline has 3 I2C HID devices. ATMEL Touchscreen,  ATMEL Touchpad and Wacom W9013 pressure sensitive.

I don't know how devices combine with these addresses. one devices have multiple addresses? can this approved I2C devices work state?

Does this help?

[image]

MrChromebox, 2021-03-11:
something changed between coreboot 4.11 and 4.12 that "broke" things, I've not had a chance to investigate unfortunately

waitlamp, 2021-03-12:
UPDATE:
I have flashed 20200317 firmware
stylus work well now
flashrom warn descriptor region and ME region read-only.
use "--ifd" option as[ their document](https://www.flashrom.org/ME) said , only flash bios region  can solve.
and strangely coreboot boot faster than 20200606 version.

MrChromebox, 2021-03-15:
ok, found the issue, just need someone to test and confirm the fix

thandlovsky, 2021-03-16:
I can flash, is this using the test OOFYFX script?

waitlamp, 2021-03-16:
My device has been sent to upgrade EMMC ( as I mentioned in an earlier issue ). I can help to test the firmware when I get it back, if you need then.

thandlovsky, 2021-03-16:
> My device has been sent to upgrade EMMC ( as I mentioned in an earlier issue ). I can help to test the firmware when I get it back, if you need then.

Can you send me an email at thandlovsky at gmail with details on how you went about getting EMMC upgraded?
I looked into doing it myself but am still unsure if the risk is worth it

MrChromebox, 2021-03-16:
@thandlovsky yes I just pushed it there for you to test

waitlamp, 2021-03-17:
> Can you send me an email at thandlovsky at gmail with details on how you went about getting EMMC upgraded?
> I looked into doing it myself but am still unsure if the risk is worth it

Soldering BGA package chip is not a easy task, 
I suggest you consider it carefully.
If you are not familiar with it, you'd better not to try it.
Component/Chips can be easily damaged by mistakes or Non-standard operation.

If you are experienced in BGA soldering, and you are determined to do that.
you can see https://github.com/MrChromebox/firmware/issues/243
I'll edit it when I get my device back.
Just wait a few days.

thandlovsky, 2021-03-17:
> @thandlovsky yes I just pushed it there for you to test
@MrChromebox Can confirm the fix updates the UEFI and the S-Pen still works.

@waitlamp I have held off on doing it myself, which is why I wondering where you found a service that would do it for you.

MrChromebox, 2021-03-17:
fixed by https://review.coreboot.org/c/coreboot/+/51507

will be included in next release

closed 2021-03-17 by MrChromebox

waitlamp, 2021-03-27:
@MrChromebox 
Surprisingly, this test firmware not only fixed the problem of the pen not working, but seems also fixed caroline CPU Speed being locked at 0.9Ghz under windows 10 (not sure it's firmware reason or intel DPTF reason)