MrChromebox/firmware #218: Acer C720 (PEPPY) Cypress Trackpad doesn't work with FreeBSD 12 and UEFI

nickegg1018, 2020-07-14:
I started with just the legacy firmware (not even the MrChromebox version, just through ChromeOS flags) and installed FreeBSD 12.1-RELEASE. Loaded the cyapa, ig4, and chromebook_platform (for Cypress, I2C, and Chromebook goodies respectively) kernel modules and the trackpad worked. 

Installed UEFI through the FW script, got everything working again except trackpad. With UEFI installed I never got a /dev/cyapa0 device for my trackpad, but dmesg would show the I2C device appearing. Running a CentOS 7 live image (I already had 7 on hand for work, I didn't try 8) the trackpad wouldn't function. Running Fedora 32 live image, the trackpad did work. So I'm guessing Linux kernel v5 has some driver baked in that Linux that v3 and the BSD kernel doesn't have. 

I reverted back to Legacy through the MrChromebox script and everything works again. Feel free to close this issue if this is more of a BSD problem and not a real firmware issue, I just noticed it only crops up with UEFI and not legacy so I know it's related to the UEFI shenanigans. This is just a machine for playing so feel free to contact me for testing.

MrChromebox, 2020-07-14:
so the only differences with the UEFI firmware really are:

* I2C devices are in ACPI mode, not PCI mode
* there are separate firmware images for PEPPY depending on the trackpad type (Elan or Cypress), because having both and not being able to dynamically enable the correct one causes Windows to have a fit

I'm guessing it's support for I2C devices in ACPI that's the issue, but I'd need to investigate further to be sure

nickegg1018, 2020-07-14:
I forgot to specify, I did go to a live Linux image and did a UEFI firmware update and selected the cypress touchpad option so I know I had the right firmware loaded for my hardware.

I'll be honest, hardware isn't my game and I know it's not your job to teach me how it works. A little bit of Googling and I at least understand the difference between ACPI and PCI now but I don't know that I'll be much help debugging that one without a little guidance on the info that would be most helpful to you.

closed 2026-11-04 by MrChromebox