MrChromebox/firmware #219: Asus Chromebox CN60 "Panther" issue with Ubuntu 18.04.4 LVM Mode

Massaguana, 2020-07-17:
Hello,

I think I've found a problem.

My Asus Chromebox CN60 "Panther" and your UEFI Full ROM seems to have problems with the Ubuntu "LVM" (Link). 

If I have installed Ubuntu 18.04.4 minimal with LVM I run into unsolvable grub problems. Grub seems to start in "emergency mode" at every system start. Anyway Grub appears and waits 30sec for input.
Since I have installed Ubuntu alone, this should not be the case.

According to my research this 30sec waiting time indicates that Grub does not get the message that the last boot OKAY was successful. So Grub appears to allow the user to use a different kernel to boot the system.

If I install Ubuntu without LVM I have no problems

Mat has asked me to open this issue

child regards

Massaguana

RolKau, 2020-12-02:
If you by LVM means the Logical Volume Manager, then this shouldn't be affected by the firmware (more than regular filesystem drivers anyway). FWIW, I have an Ubuntu installation (version 20.04 though) where the disk is partitioned with LVM, and Grub does not enter failsafe mode. What does the command `sudo parted /dev/sda unit s print` give on your system? (It will print the partition table of the first harddisk, measured in number of sectors)

Massaguana, 2020-12-03:
here is the result
```

Modell: ATA TS256GMTS430S (scsi)
Festplatte  /dev/sda:  500118192s
Sektorgröße (logisch/physisch): 512B/512B
Partitionstabelle: gpt
Disk-Flags: 

Nummer  Anfang    Ende        Größe       Dateisystem  Name                  Flags
 1      2048s     1050623s    1048576s    fat32        EFI System Partition  boot, esp
 2      1050624s  500117503s  499066880s  ext4
```

RolKau, 2020-12-07:
Your partition table looks normal (there is no overlap between the partitions), but at the same time there is no LVM partition there either, so I guess this is after the new installation without LVM. I am afraid you won't get any answers to what went wrong until another running system exhibits a similar error.

Massaguana, 2020-12-07:
Matt asked me to do this post... I wasn't really expecting a solution...

Since I don't see any advantage in LVM it's okay.

closed 2026-11-04 by MrChromebox