MrChromebox/firmware #220: issue with booting from bios

tcavlee, 2020-07-23:
I am having issues with my Asus Chromebox 3 not booting from bios to loaded program. I have to plug in a different keyboard and unplug power multiple times while hitting the esc button F12 button enter button or control alt delete before I can get into the boot menu where I can change the boot order to the hard drive. Once I do this the program loads. I have had this problem with Gallium, Ubuntu, Manjaro, Linux lite, Linux cinnamon and a couple of other distros that i can't remember.
Here is what I get for the system:
Operating System: Manjaro Linux
KDE Plasma Version: 5.19.3
KDE Frameworks Version: 5.72.0
Qt Version: 5.15.0
Kernel Version: 5.6.19-2-MANJARO
OS Type: 64-bit
Processors: 4 × Intel® Core™ i3-7100U CPU @ 2.40GHz
Memory: 15.5 GiB of RAM
Graphics Processor: Mesa Intel® HD Graphics 620

I have updated the firmware and I am running UEFI from MrChromebox.

tcavlee, 2020-07-27:
So I continue to fight everyday for 15 to 20 minutes to get past the bios boot. The screen flickers in and out with the Rabbit logo teared across the screen. I try esc, enter, control alt Del, f12, f10, f2 sometimes all at the same time to no effect. The screen just goes black and nothing further happens. I have to unplug the power and start all over again until something finally takes and the Rabbit shows up like it is supposed to and my Linux distro loads. Mr. Chromebox any help would be appreciated or anyone else's help! @MrChromebox

MrChromebox, 2020-07-27:
there's a single key, ESC, to enter the UEFI boot menu/settings. Hitting any other key is just flooding the keyboard buffer with garbage, so don't bother.

You have the latest firmware loaded? what type of connection to your display? what type/resolution display?

tcavlee, 2020-07-27:
@MrChromebox    So I have tried to use esc when the rabbit shows up and it doesn't always take me to the boot screen. I am using an Asus VP278 via HDMI. I have run the update to insure I have the latest firmware and the resolution is 1920X1080 refresh rate 60Hz. Most of the time now it never actually takes me to the boot screen where I can change the boot order. Once it decides that it likes the key combo I have used the bios works like it should. The rabbit shows up the directions to use escape to enter bios or boot shows up and then the distro loads like it should.

tcavlee, 2020-07-27:
@MrChromebox I have used your firmware to load Linux on a Samsung Chromebook Pro without any issues. It is running Gallium but I have had the issues since I have changed my Asus Chromebox 3. Like I said I have tried multiple distros to no effect. I have settled on Manjaro because I have found I like it. I am working on learning Linux, programing and such because I find it really interesting and fascinating. I guess that is part of my frustration here is I don't know enough or understand enough to give you better information. Sorry

closed 2020-08-04 by tcavlee