MrChromebox/firmware #222: The device you inserted does not contain ChromeOS

hh-here, 2020-07-25:
The device you inserted does not contain ChromeOS
Model Snappy E25-C3C-N2Y-A7Z-U3Q-A2L
hp 14-ca043cl {ApolloLake}
GalliumOS (chrx) on 18Gb internal partition

1.	I'm pretty happy to have this, more-or-less working.
2.	Thanks VERY much.
3.	If/when I walk away from the computer and it 'suspends' or whatever it has defaulted to do:
	I come back to it soon after and can log back in with the chrx password.
	If I leave it for a longer while, when I 'wake it up' by pressing the space-bar, it says:
	"The device you inserted does not contain ChromeOS
	Model Snappy E25-C3C-N2Y-A7Z-U3Q-A2L"
	Anything I do leads to rebooting with the only option being to go back to the chromeOS.
	From there I am able to run the 'linux' script to install/update the firmware ==>> OK
	YES! I can 'get around this' by telling the computer to stay awake.
	Each of those options is inelegant.
	
4.	Does this seem to be a firmware issue or a user issue?
5.	It uses 20180912.bin Should I use a more recent version?
6.	Is there a more elegant solution than those I mention above?

Looking forward to advice/suggestions/guidance . . .

MrChromebox, 2020-07-25:
3) this is a known bug with legacy boot mode on all CR50 (2017+) ChromeOS devices. The only workaround is to disable sleep/suspend. A newer kernel compiled with the CR50 TPM driver which should not have this issue

4) stock firmware issue

5) RW_LEGACY firmware version is not a factor

6) using my script to set the firmware boot flags will prevent you from needing to re-set the dev_boot_legacy flag if the system goes to sleep without the TPM telling the CR50 that boot was successful

closed 2020-07-25 by MrChromebox

hh-here, 2020-07-26:
Thanks very much for this heads-up/clarification.