MrChromebox/firmware #225: boot from SD card slot

peepo, 2020-08-13:
enhancement request: boot from SD card slot

moved my asus c200m to single boot Gallium some years ago.

decided to upgrade, installed latest firmware tx!

SD card in slot is listed in the Boot Menu
but boot does not progress beyond 4 lines
ERROR PPS state mismatch... 

boots from usb with SD card holder fine!

not clear to me why boot from SD card slot fails.

tx

theflyingape, 2022-02-12:
Stock ChromeOS w/RW_LEGACY: ASUS Chrombox 3 Celeron shows in boot menu and boots from SD card.  Sadly, the SD card does not present itself on ASUS Chromebox 4 i7 and therefore no joy there.

MrChromebox, 2022-02-13:
@theflyingape please start your own issue as you're using a completely different device. And be sure to report what firmware type/version is being used

closed 2026-11-04 by MrChromebox