MrChromebox/firmware #226: syntax error

spicepigs, 2020-08-29:
acer c710-2847
for some reason---because of the gfw,
i cant reach the /mrchromebox.tech to get the firm,
it always be conect error.
> cd ~; curl -L -O http://mrchromebox.tech/firmware-util.sh; sudo bash firmware-util.sh

so i change the url to the github ,
> cd ~; curl -L -O https://github.com/MrChromebox/scripts/blob/master/firmware-util.sh; sudo bash firmware-util.sh

it connect fine ,the firm was downloaded and i run the bash command ,i got an error code:

syntax error on line 6

what happened,what should i do?

MrChromebox, 2020-08-29:
> so i change the url to the github 
>> cd ~; curl -L -O https://github.com/MrChromebox/scripts/blob/master/firmware-util.sh; sudo bash firmware-util.sh

that's not the correct URL, you downloaded a github webpage, not the script. Did you bother opening the file to see what it was?

you can try:
https://github.com/MrChromebox/scripts/raw/master/firmware-util.sh

closed 2020-08-29 by MrChromebox

spicepigs, 2020-08-29:
thank you, it works,starting download now~