MrChromebox/firmware #227: Stuck after setting boot order with firmware-util.sh

Kema-dev, 2020-09-09:
Hello there !

I'm pretty much a newbie to GitHub, I hope I'm posting at the right place.

I bought an Acer Chromebook 315 a few days ago, and wanted to dual boot ChromeOS / Linux. I made my bootable USB key through Windows 7 on another PC, and when i try to boot on it from this other PC it works as expected.

The problem is that I wanted to enable USB boot on my Chromebook, so I followed the instruction to use Chrome OS Firmware Utility Script. I removed the battery then removed WP and started the script. I successfully changed boot option into "1 sec timeout chrome os", then reboot and everything worked as expected. Then, I changed to "1 sec timeout legacy" and reboot, but now I keep having the dev menu (bitmap) on startup, then it tries to boot but just freezes (I can't change the language after 1 sec but I can't before the timeout, wich I guess I shouldn't have set to 1 sec). I tried pressing Ctrl + L to get more boot options before the freeze, but i just have the choice marked "2 TianoCore     TianoCore bootloader", and when i try to launch it it just freezes as it does if i wait 1 sec in dev bitmap.

The fact is I had ChromeOS running well before I change the boot order, it makes me think that ChromeOS is still alive somewhere but I just can't reach it... I didn't re-enable WP.

Before using the utility script, TianoCore didn't want to boot either when I tried using Ctrl+L method.

Do I need to flash my Firmware using a USB Programmer like this one ?

https://www.amazon.fr/Programmateur-CH341A-adapt%C3%A9-adaptateur-conversion/dp/B07TXGJPWW/ref=sr_1_1_sspa?__mk_fr_FR=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=CH341a+USB+flash+programmer&qid=1599676772&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExTEE2TkNMSlAxN1lCJmVuY3J5cHRlZElkPUEwOTkwMTA3MjRaSTAwTDhSQTc4RyZlbmNyeXB0ZWRBZElkPUExMDMzNTQ1MjZXRENBU0Q1OTRKQiZ3aWRnZXROYW1lPXNwX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=

Thanks !

MrChromebox, 2020-09-09:
> The problem is that I wanted to enable USB boot on my Chromebook, so I followed the instruction to use Chrome OS Firmware Utility Script.

your device is not supported / doesn't have RW_LEGACY firmware. You can't boot anything other than ChromeOS until Google fixes things on their end unfortunately.

> The fact is I had ChromeOS running well before I change the boot order, it makes me think that ChromeOS is still alive somewhere but I just can't reach it

sure you can, just use CTRL+D

> Before using the utility script, TianoCore didn't want to boot either when I tried using Ctrl+L method.

because it's broken on Google's end

> Do I need to flash my Firmware using a USB Programmer like this one ?

absolutely not, just use CTRL+D to boot ChromeOS and then re-set the boot options back to ChromeOS default (or factory default if going to exit Developer Mode)

Kema-dev, 2020-09-09:
Hem yes, I was sure I already tried but yeah I actually just had to press Ctrl+D haha, my bad !
Then I'm going to try out some VM solution such as Crouton to use a proper Linux distro, thank you !

closed 2020-09-09 by Kema-dev