MrChromebox/firmware #228: Fix vt-x/vmx for devices supporting it?

Matthew-Jenkins, 2020-09-24:
Howdy, 

I've noticed that vmx is disabled on my i3 ganolf which otherwise has vmx support. I would imagine this is disabled on other devices as well. Is there a way to enable this in a new build?

Thanks,
Matthew

MrChromebox, 2020-09-24:
VT-x (vmx) has been enabled in the UEFI firmware on every device from the very start. VT-d is enabled on CPUs that support it. I'm not sure what you're checking that's reporting vmx is not enabled, but if you're running my UEFI firmware it's incorrect. If you're running RW_LEGACY, then that's still the stock firmware at its core (RW_LEGACY is just a bootloader), and nothing can be done to enable vmx there.

Matthew-Jenkins, 2020-09-26:
I'm running Fedora 32 Workstation. I've been using with Fedora since shortly after I bought it about 5 years ago, thanks to the firmware you've provided. However I've been tweaking it and working on it to make it work like it did when it was new and rechecking for optimizations. Reviewing my zswap was a big one. Disabling spectre and meltdown fixes was another one that really sapped performance. 

But I notice that for the iTLB multihit it says VMX is disabled and I don't see vt-x/vt-d listed as an extension available. See here:
https://www.kernel.org/doc/html/latest/admin-guide/hw-vuln/multihit.html VMX Disabled means that vt-x/vt-d is turned off or not present so there's no mitigation required.

While here https://ark.intel.com/content/www/us/en/ark/products/84698/intel-core-i3-5015u-processor-3m-cache-2-10-ghz.html  it states that vt-x and vt-d are indeed present on the chip. So it isn't being turned off by Linux, so I can only think the firmware?

MrChromebox, 2020-09-26:
can you please tell me what type of firmware and version you are running?

Matthew-Jenkins, 2020-09-27:
Is this it?
 ChromeOS Device Firmware Utility Script [2020-07-18] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**   Device: Toshiba Chromebook 2 (2015) CB30/CB35 (GANDOF)
** Platform: Intel Broadwell
**  Fw Type: Full ROM / UEFI
**   Fw Ver: MrChromebox-4.12 (06/04/2020)
**    Fw WP: Disabled
*********************************************************

MrChromebox, 2020-09-27:
VT-x and VT-d are both enabled in the 2020-06-04 Broadwell firmware for all devices; I just confirmed on an i7-5500u here. 

The easiest way to check for VT-x/vmx is `cat /proc/cpuinfo | grep vmx`

I booted a Mint 20 live USB (kernel 5.4) and `cat /sys/devices/system/cpu/vulnerabilities/itlb_multihit` returns `KVM: Mitigation: Split huge pages`

closed 2024-04-28 by MrChromebox