MrChromebox/firmware #231: Link (Chromebook Pixel 2013) does not resume when open lid from suspend mode on Linux

gonzalomarcote, 2020-10-08:
Using latest version (updated with ChromeOS Firmware Utility Script) and Linux (Debian 10) on Link (Chromebook Pixel 2013) when the system is suspended and you open the lid, system doesn't start (resume) automatically. You have to press power button.
This does not happen when the sistem is complete powered off, because the system starts properly. Only when it is suspended.

MrChromebox, 2020-10-08:
latest UEFI firmware (2020.06.04) has no issues resuming from suspend when opening the lid, regardless of whether the system was suspended via lid close or manually prior to. Just tested here on Mint 20.

coolstar, 2023-06-30:
Suspend/Resume works fine on Link here on both Manjaro and Win10 22H2

closed 2023-06-30 by MrChromebox