MrChromebox/firmware #233: Dell Latitude 5300 2-in-1

crofh, 2020-10-26:
Any chance of adding full support for this one in the near future?

Would it even be possible to somehow update the firmware/BIOS with this one?  

https://www.dell.com/support/home/en-ca/drivers/driversdetails?driverid=f91gn&oscode=wt64a&productcode=latitude-13-5300-2-in-1-laptop

It looks like they used the same hardware and just turned it into a Chromebook product, as the Windows version has been discontinued.

Pardon me if this is a stupid question.

Thank you.

MrChromebox, 2020-10-26:
> Any chance of adding full support for this one in the near future?

the 5300/5400 don't support flashing via CCD (Suzy-Q cable), and don't have a clippable chip (they use a WSON-8), so they'd be quite difficult to unbrick should something go wrong.

> It looks like they used the same hardware and just turned it into a Chromebook product, as the Windows version has been discontinued.

no quite, you couldn't just flash firmware between the two, as there are actual hardware/board differences

I did just release legacy boot firmware for these devices with UEFI, so should allow you to run another OS

acastro60, 2020-11-05:
I appreciate the response.  I was able to install Windows 10 on it in Legacy mode, but a bunch of drivers are missing or unknown.  I tried downloading all the Windows 10 drivers for the 5300 2-in-1 non-chromebook, but only managed to install a couple before it started freezing up or hanging during the driver installation.  There are also some unknown drivers related to the Chrome BIOS/Firmware.  I was able to install ubuntu 20.04 LTS without issue so that's something at least.  Thanks for all of your efforts.

MrChromebox, 2020-11-05:
Windows doesn't have drivers for devices which are unique to Chromebooks, unless someone writes them -- that's just the way it is. Stock Google firmware also isn't tested for Windows compatibility, and often has issues, so not surprising you ran into problems.

closed 2026-11-04 by MrChromebox