MrChromebox/firmware #235: CN62 (ChromeBox2) Running Memtest86, CPU temperature gets to 103c when running.

FingerlessGlov3s, 2020-11-18:
Hi Guys,

I've got a CN62 (ChromeBox2) running Memtest86 so I can test some new ram in the device, the CPU gets to 103C when running.
If I remove the board from the case the CPU temperature drops to the 60-70 range, which is a big difference. It seems to me as if the FAN curve profile isn't happening when running memtest86 software, as the fan speed isn't changing at all, its sounds and feels to be at a fixed RPM.

Is this an issue or know bug with certain OSes/Utilities?

memtest86: https://www.memtest86.com/

FingerlessGlov3s, 2020-11-18:
Soon as I booted Linux, the fan went full speed to cool down the CPU. As you can see from the below output the CPU is now 45C. So the curve does work.

sensors output
```
# sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +45.0°C  (high = +105.0°C, crit = +105.0°C)
Core 0:        +44.0°C  (high = +105.0°C, crit = +105.0°C)
Core 1:        +44.0°C  (high = +105.0°C, crit = +105.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +46.0°C  (crit = +104.0°C)

pch_wildcat_point-virtual-0
Adapter: Virtual device
temp1:        +97.0°C
```

MrChromebox, 2020-11-18:
all of the 2nd/4th/5th-gen Chromeboxes use an IT8782F SuperIO chip for fan control, which is programmed by coreboot for manual (OS) fan control via ACPI. There's a default fan speed before the OS takes over fan control, but it's low enough to not be audible. If you're booting memtest from grub, then there's no ACPI fan control, and the fan speed will stay at the default (low) level

FingerlessGlov3s, 2020-11-18:
That sounds about right, the CPU stays cooler when under Linux. Doing a stress test on all threads, the CPU stays under 70°C

Although the PCH with lm-sensors reports pretty high, so I re-pasted the CPU and PCH, same temps as last time, wonder if the tool reports it incorrectly, I would of thought 115°C PCH would be too hot?

```
 sensors && uptime                                                                                                                                                                                   pvehost: Wed Nov 18 19:24:34 2020

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +66.0°C  (high = +105.0°C, crit = +105.0°C)
Core 0:        +66.0°C  (high = +105.0°C, crit = +105.0°C)
Core 1:        +66.0°C  (high = +105.0°C, crit = +105.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +67.0°C  (crit = +104.0°C)

pch_wildcat_point-virtual-0
Adapter: Virtual device
temp1:       +115.0°C

 19:24:34 up 23 min,  1 user,  load average: 4.21, 4.06, 3.13
```

MrChromebox, 2020-12-09:
yeah there's no way that 115*C is correct, the PCH is on the same die as the CPU and GPU, so they should all be within 1-2*C of each other

FingerlessGlov3s, 2020-12-09:
> yeah there's no way that 115_C is correct, the PCH is on the same die as the CPU and GPU, so they should all be within 1-2_C of each other

This is what I suspected, but there are two die's on the CPU package. But from what your staying one of the die's is CPU, GPU and PCH. Not sure what the other die is for though.

MrChromebox, 2020-12-09:
I believe the CPU+PCH is on one die, and the GPU on another. All are on the same package.

closed 2026-11-04 by MrChromebox