MrChromebox/firmware #236: ChromeOS won't repair itself after flashing the default firmware.

higgy999, 2020-11-18:
When restoring my Chromebook to it's default firmware and trying to repair ChromeOS with a flash drive (I followed Google's instructions) it keeps saying it failed with "An unexpected error has occurred." I have a Acer Chromebook 11 CB3-111 (Gnawty). It's not completely bricked it still boots into that state of needing a repair and the developer enabled screen. Can anyone help me restore ChromeOS or figure out how to change the enable USB botting flag?

MrChromebox, 2020-11-19:
pop the USB into another machine and retrieve the recovery.log, and throw it on pastebin.

setting the dev_boot_legacy flag won't help since there's no RW_LEGACY firmware to boot -- you've need a custom recovery to install that and set the flag

higgy999, 2020-11-20:
I found it in the tmp partition. The file says the name of a folder in the same directory. In there there is 5 files. dmesg, hardware_diagonstics.log, progress.log, recovery.log, spinner.log

This is the recovery.log in the folder the other log pointed to: https://pastebin.com/HXe2R9SH

MrChromebox, 2020-11-21:
seems like Google broke something in Recovery as part of a recent update. The recovery process trying to update the firmware despite both the RO and RW version strings matching, and further is trying to erase/write regions that are marked as locked. It should not be doing either of those things. I've reached out to one of my Google contacts to see if they know anything about it

MrChromebox, 2020-11-21:
I'm told this might be an issue with the current R86 images. Please try using an R85 (or R84) image for your device from the link below and then updating after a successful recovery. LMK how it goes.

https://cros-updates-serving.appspot.com/

higgy999, 2020-11-22:
I used R85 and successfully recovered! Thank you so much, I thought this laptop was doomed. I was also able to update to 86.

closed 2020-11-22 by higgy999