MrChromebox/firmware #237: "alloc magic broken" error booting OS

SuperSamus68, 2020-11-19:
Failing to load OS. Tried multiple USBs running GalliumOS-3.1 Broadwell isos packaged with Rufus

"error: unknown filesystem.
alloc magic is broken at 0x74ec9880: 74dee140
Aborted. Press any key to exit."

Hardware is Google Pixel Chromebook 2015 C1501W. WP pin is removed.
FW: MrChromebox-4.12 06/04/2020
Intel (R) Core (TM) i7-5500U CPU @ 2.40GHz
16384 RAM
 
Installing UEFI (Full ROM) Firmware went smoothly. Rebooted. Hitting the error when I boot from hard drive or a flash drive.
Rufus isos did work pre-firmware installation.

Any ideas how to fix this? Thanks.

MrChromebox, 2020-11-19:
https://www.reddit.com/r/GalliumOS/search?q=alloc+magic+broken&restrict_sr=on&sort=relevance&t=all

TLDR: you didn't write the ISO to USB properly.  Use Rufus in dd mode, not ISO mode.

SuperSamus68, 2020-11-19:
That did it. Thank you.
Apologies for the common question. DuckDuckGo couldn’t yield any results.

Future readers:
Select the .iso in Rufus. After hitting START it’ll prompt you between modes.

closed 2020-11-19 by SuperSamus68

FavroiteGamers, 2022-12-10:
Thanks this helped me as well!