MrChromebox/firmware #239: HP Chromebox CB1 Haswell issues

xbipin, 2020-11-23:
i have a few HP Chromebox CB1 Zako with Haswell core i7 processors with windows 10 pro and sufferring a few issues with 4.12 bios release:

These boxes all have the Atheros Qualcomm Ar9462 Ar5b22 wifi/bluetooth card and as soon as thats plugged in and you boot to windows, there is a flood of WHEA logger warning messages as below in event viewer and it just doesnt stop and with some time the machine cuts power and reboots
A corrected hardware error has occurred.

Component: PCI Express Endpoint
Error Source: Advanced Error Reporting (PCI Express)

Primary Bus:Device:Function: 0x2:0x0:0x0
Secondary Bus:Device:Function: 0x0:0x0:0x0
Primary Device Name:PCI\VEN_168C&DEV_0034&SUBSYS_1864103C&REV_01
Secondary Device Name:

The other issue is wireless mouse also dont appear as smooth, morover they lag like hell due to the above

The CPU temprature on all the boxes touches 100+ degrees very quickly with some load, the fan isnt audible but i can feel it pumping hot air out the back grill so either its not spinning fast enough or something else

MrChromebox, 2020-11-23:
the AER error is spurious, will be fixed in the upcoming 4.13 release.

Haswell and Broadwell Chromeboxes require the use of Speedfan to have automatic fan control under Windows, because Windows doesn't like the firmware ACPI for the fan for some reason.

xbipin, 2020-11-23:
ok thanls for that info, is there a beta firmware i can try for 4.13?

MrChromebox, 2020-11-23:
no, coreboot 4.13 was released all of 3 days ago; I've not begun preparing a new release.

xbipin, 2020-11-24:
so will there be a fix for the fan control on windows or we just need to keep using speedfan with these boxes? if its less likely to get fixed then why not keep the fan at 50% by default in bios and if the OS controls it then good or else atleast the processor doesnt go thermonuclear.

MrChromebox, 2020-11-24:
I try not to do things in firmware to work around Windows' idiocy when they negatively impact other users. If I were to set the fan to default to 50% like you suggest, there would be a loud roar of the fan at startup, for a good 10s or so until Linux took over things. I'm not going to do that. If someone wants to debug the fan ACPI and figure out why Windows is being stupid, then I'll gladly implement a proper fix.

xbipin, 2020-11-24:
ok i understand that and its fair enough, eagerly waiting for 4.13 so can get of those WHEA error flood

MrChromebox, 2020-11-24:
shoot me an email at mrchromebox at gmail and I'll give you a test build to try

xbipin, 2020-11-25:
Sent

xbipin, 2020-11-26:
yesterday i ran latencymon tool in windows to figure out mouse stutter issues and after around 5-6mins of it running it reports that the cpu is being throttled even though speedfan is running and the cpu cores are under 50 degrees C. Then i ran HWinfo tool which also reports core power limit exceeded and package/ring power limit exceeded even though its mostly just a clean windows install.

MrChromebox, 2020-11-28:
I tested latencymon and hwinfo here and seeing the same issues with latencymon, but nothing related to CPU throttling. Additionally, latencymon is eating up quite a bit of CPU time, the % which keeps growing until it reports latency issues, so I'm not sure that it's a good indication of any actual issue.

That said, I'm not seeing the issue under a 4.13 test build (unless the system goes to sleep) so who knows. I'll have you update to the 4.13 release (once it's out) and we can troubleshoot if the issue persists

xbipin, 2020-11-29:
One other issue I noticed in windows is, once booted and on desktop and you plug a USB flash drive, it works fine but if you leave it inserted and reboot windows then it stops working, device manager shows unknown device or something, the only way to get it to work again is unplug and replug it after which it gets back to normal.

Im guessing wireless mouse stutter could be related to it because if I unplug the dongle and replug then it does reduce to certain extent.

MrChromebox, 2020-11-29:
I'm not able to reproduce that here, no such issues with mine, using a
Logitech unifying dongle

xbipin, 2020-12-04:
wireless mouse stutter i face on almost all mice, tried lenovo, hp, logitech
the USB flash drive disappearing on reboot i tried with kingston, sandisk, lexar
im using the 4.12 bios release and the latest build of windows 10

xbipin, 2021-03-31:
any idea when 4.13 will be out?

MrChromebox, 2021-03-31:
whenever I have time to prep and test another release

MrChromebox, 2021-04-26:
fixed in 4.13 release

closed 2021-04-26 by MrChromebox