MrChromebox/firmware #240: CXI3 Kabylake 4K display garbled in text mode

RolKau, 2020-11-23:
I have an Acer Chromebox CXI3 (SION); the version that has an i3 processor, Kabylake chipset and Intel UHD Graphics 620 (rev 07).

Installing new firmware was just a matter of following the recipe (removing write protect screw, booting in Developer Mode, download and running the firmware-util script) and went about without any hitches. The firmware-util script describes the original firmware as "Google_Fizz 10139.163.0 (02/21/2019)", and the new firmware that was downloaded was "coreboot_tiano-fizz-mrchromebox_20200604".

However, when it now boots, text mode is garbled, reminiscent of the DOS days when a program attempted to write to text mode while the display was stuck in graphics mode. It is as if all the characters have collapsed into little pixel chunks. For instance, the attached screenshot shows the text "Press any key to continue". Knowing that, you can see that the relative length of each word matches. In the beginning, all the text are at the bottom of the screen, but it later turned out the the entire screen could be eventually be filled.

This happens both when using HDMI as well as when using DisplayPort through an USB-C adapter, both connected to a 4k display. I tried using an old 1600x1200 monitor through yet an DisplayPort-to-DVI adapter, but then the screen went entirely black; I am not quite sure if this is because everything now was outside the range of the screen or if nothing is actually shown. I don't have any lower resolution monitors with native HDMI or DP connectors, unfortunately.

Realizing that what I saw was the firmware complaining about not being able to boot the old ChromeOS installation and then dropping me into an UEFI shell, I was able to boot an USB stick, for instance Ubuntu 20.04, using the standard command:

    fs0:\EFI\boot\bootx64.efi

Once the installer goes into graphical mode, the display is fine again, and disk, graphics, keyboard, mouse and network seems to be fully functional. However, firmware, GRUB boot loader and text mode is still garbled.

Interestingly, SystemRescueCD (which is Arch-based) does some mode change upon startup and is able to show text mode from very early in the boot. Therefore, it seems that what is wrong is the configuration of the default text mode, in some way.

MrChromebox, 2020-11-23:
I'm not seeing this issue with WUKONG on the 2020.06.04 firmware, on either HDMI or USB-C->DP/HDMI, connected to a 1440p display. I don't have a 4K display handy to test though.

RolKau, 2020-11-24:
I think it may be related to the resolution: If I edit `/etc/default/grub` to have the option

    GRUB_GFXMODE=1024x768

and then run `sudo upgrade-grub2` (being explicit here in case anyone else comes along with the same problem and needs quick instructions), then the Grub bootloader is functional again. Several other modes work as well, but native 3840x2160 does not. Incidentially, only the video modes with a 4:3 aspect ratio seems to work.

At one point, I could even enter the firmware from the "UEFI Fireware Settings" option in Grub and have it inherit the display settings so that everything worked normally. Alas, after some more tinkering I am unable to reproduce that condition, even after restoring all options to what they were at the time.

Would it be of interest that I tried to dump the display panel timing parameters and/or the Video BIOS Table from the stock ROM?

MrChromebox, 2020-11-24:
sounds like grub is the problem then, not the firmware? have you tried setting the gfx_mode to auto? or keep?

the VBT is used at display init by the firmware, and there appears to be no issue with that. The same VBT is used for the UEFI firmware as the stock firmware

RolKau, 2020-11-24:
> sounds like grub is the problem then, not the firmware?

No, even the coreboot startup logo is garbled. The image I attached above is from the firmware boot menu, before grub or any OS was installed. On the contrary, the display is garbled until something initializes it. The OS can set it to 3840x2160 just fine. Grub cannot, using the efi_gop backend.

MrChromebox, 2020-11-24:
ok, hooked up my box to a TCL 4K TV using a USB-C to HDMI 2.0 adapter, and no issues displaying the boot logo, UEFI settings, or grub at 4Kp60.

I'd just keep using your current workaround of setting grub to a standard VESA resolution

RolKau, 2020-11-24:
Encouraged by your results, I decided to check out how the Sion fared with the various monitors around in the manor:

* an [AOC U2790PQU](https://www.displayspecifications.com/en/model/8c801a73), identical to the original monitor but acquired at a different time (to rule out hardware errors in the original)
* a [Philips 328P6V](https://www.displayspecifications.com/en/model/2c3381e)
* a [Samsung U32H850](https://www.displayspecifications.com/en/model/6e93c35), known to be "funky" in that I had to configure it using a manual modeline to get 4K to work togethter with an Nvidia graphics card earlier
* a [Sony KD49x8305C](https://www.displayspecifications.com/en/model/254d125) TV, to see if I could get the same results as you there

All monitors were connected using a cable directly from the HDMI output of the Chromebox to the monitor, to rule out any adapter interference. (If I attempt to use a USBC -> DP -> HDMI chain of adapters, the TV doesn't show any signal, akin to when I tried a similar setup with DVI to an older monitor earlier). I am of course aware that the [Intel UHD 620 Graphics](https://www.intel.com/content/www/us/en/support/products/126789/graphics/graphics-for-8th-generation-intel-processors/intel-uhd-graphics-620.html) chip does not support HDMI 2.0, but the TV does not have a DisplayPort input.

Entering the Grub command-line, I first listed the EFI GOP video modes with the command `videoinfo` before I booted to a SystemRescueCD USB stick using the Grub commands:

    chainloader (cd0,msdos2)/efi/boot/bootx64.efi
    boot

(I reckon you already know this; it is for the poor souls rumaging through issues to find something they can use; I myself have been saved by this on numerous occations).

The AOC and the Philips gave the same results as before, eliminating the chance of everything just being peculiarities of the original monitor. Both reports a 3840x2160 video mode as preferred.

The Samsung showed only a blank screen in the firmware phase, but came to life when Grub changed to the 1024x768 VESA mode. Curiously, this monitor reports that a 3840x2160 video mode is preferred, but it is not in the list. I did mention it was funky, didn't I? Booting further to SystemRescueCD, it enters a 4K mode, though.

The really interesting part starts with the Sony TV: It shows the Coreboot rabbit logo upon startup, and when pressing ESC the firmware boot menu shows just fine. However, the TV is now in HD mode (1920x1080). In Grub, `videolist` shows a bunch of VESA modes and a 1920x1080 mode, with the latter being preferred. There is no 3840x2160 GOP mode. SystemRescueCD brings the X server up in 1920x1080 mode, but I can manually change this to 3840x2160 in the settings panel, and that work fine.

Dumping the monitor EDID data with

    sudo cat /sys/devices/pci0000\:00/0000\:00\:02.0/drm/card0/card0-HDMI-A-1 | parse-edid

indeed reveals that the AOC is close to CVT-Reduced timings, the Samsung has a "strange" 4K mode (meaning that it is neither CVT nor GTF, but have slightly larger clocks), and that the Sony doesn't even list its 4K mode *and* it has the same funny timings as the Samsung, but still works best.

[aoc-u2790pqu.edid.txt](https://github.com/MrChromebox/firmware/files/5592808/aoc-u2790pqu.edid.txt)
[samsung-u32h850.edid.txt](https://github.com/MrChromebox/firmware/files/5592810/samsung-u32h850.edid.txt)
[sony-kd49x8305c.edid.txt](https://github.com/MrChromebox/firmware/files/5592811/sony-kd49x8305c.edid.txt)

I hesitate to inquire, but it would have been very interesting to see the EDID output of your TCL screen, to compare.

MrChromebox, 2020-11-25:
I'll try to pull the EDID from it in the next few days

MrChromebox, 2021-01-10:
@RolKau I had someone else email me with a similar issue, and an updated firmware build fixed it for them. Shoot me an email at mrchromebox @ gmail if you'd like to give it a try

RolKau, 2021-01-12:
Having tried this version (`coreboot_tiano-fizz-mrchromebox_20210103.rom`), I can report that it fixed it for my computer as well. Flashing the new version was problem-free; I didn't even have to add the internal disk to the boot menu again. It now shows the bunny logo on startup, I can enter the boot menu or the UEFI shell, and, having in my `/etc/default/grub` the lines:

    GRUB_GFXMODE=3840x2160x32
    GRUB_GFXPAYLOAD_LINUX=keep
    GRUB_VIDEO_BACKEND="efi_gop"

I get a full-screen Grub menu; from its command-line the `videoinfo` command now shows the mode 3840x2160x32 being both available and preferred.

closed 2023-02-03 by MrChromebox