MrChromebox/firmware #241: Lenovo x131e Cromebook : Lid Wifi LED and other OS issues.

agnivo007, 2020-11-26:
The wifi light on back of the lid of Lenovo x131e chromebook does not light up even if wifi and bt are active.
It functioned properly on stock BIOS and firmware+ChromeOS 69.

Flashed to full UEFI firmware using the script and installed Cloudready. Noticed the green LED stopped lighting up. It just lights up for a split second while powered on and then never seen.

Tried various live boot distros (eg-MX Linux) as well as GalliumOS 3.1 sandybridge ISO, same result.

Also, tried installing GalliumOS on the HDD (SSD has cloudready running) several times but it never completes. Gives an error installation failed and crash data sent to developers. Anything wrong with the BIOS?

mrChromebox 4.12 UEFI BIOS
Cloudready 85.4 Home stable.

[image]

MrChromebox, 2020-11-26:
not sure why the Wifi LED wouldn't work as before, that's controlled by the EC not coreboot. But same here on my x131e.

GalliumOS install issue is likely due to being connected to wifi and it pulling down an updated package that breaks something; has nothing to do with the firmware.

I'll leave this issue open for the Wifi LED, but it's not something I intend to spend any time trying to fix

agnivo007, 2020-11-26:
Well, I'll try with the OS later as you've suggested.

However, my 5yr old son was used to the greenie light and loved it. As chrome OS updates stuck at v69, I decided to do full UEFI update and install cloudready. Now, I have nothing to answer to a kid who lost his greenie...you know kids! I understand you are busy with serious stuff, still it'd be nice if you can do anything regarding this issue. My little kid would be thankful...

agnivo007, 2020-11-26:
I've also tried clearing the NVRAM, no result as expected.

agnivo007, 2020-11-27:
OK, GalliumOS installed fine this time without wifi connected, trying windows 8.1 now, will be fine too I hope.

Regards to the LED issue, I read up the value of /sys leds of ath9k-phy1, brightness was = 255, if that means something. So, software is instructing LED to glow, but it doesn't light up, but only for a split second when I cold start the laptop.

Should I try reflashing the UEFI firmware again?

MrChromebox, 2020-11-27:
> Should I try reflashing the UEFI firmware again?

no, that will have zero impact.

I did take a quick look yesterday and compared the EC setup/config of the stock firmware to my UEFI and don't see any differences (nor anything potentially related)

agnivo007, 2020-11-27:
So, you too have the LED light issue and relate nothing to the EC...

BTW, Installed Win8.1 (updated to latest), it says no battery is detected (battery cross icon in taskbar) anything related to UEFI?
Device manager shows acpi battery driver loaded fine.

MrChromebox, 2020-11-27:
Win 8.1 isn't supported, YMMV. Battery under Win 10 works properly (as tested here)

Dev727, 2021-01-07:
I would like to confirm that my WiFi LED is also not working. Seems to be a consistent issue. 

I would be willing to take a look at the EC, and see if I can figure anything. Not sure if the EC is open though.

MrChromebox, 2021-01-07:
the EC isn't open. If someone wants to poke around, great, I gladly accept PRs :)

closed 2026-11-04 by MrChromebox