MrChromebox/firmware #242: ASUS Chromebox 3 (CN65) Crashes under high CPU load

lanrat, 2020-11-30:
I'm not sure if this is the right place to file this issue, or even if its Firmware related, so feel free to close if this is not the correct place.

Device: ASUS Chromebox 3 (CN65) Fw Ver: MrChromebox-4.12 (06/04/2020)

I'm running Debian Linux, and whenever I run any process with high CPU load the CN65 instantly locks up and is entirely unresponsive, needing a hard reboot to come back to life. I've had this happen with multiple different processes, all that push the CPU. Unfortunately as it happens the instant the CPU load gets to high I'm unable to see anything in the system logs, and am currently at a loss on how to debug.

I'm using the stock 90W power supply.

MrChromebox, 2020-11-30:
which CPU? any way to reliably reproduce? If so, I'd try booting a live USB and replicating there to rule out an issue with your install/kernel etc

lanrat, 2020-12-01:
The CPU is a Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz.

In the past this has happened whenever I run aircrack-ng, hashcat, or gunzip. I just tried running the programs again and it appears to not crash. I'll run some more tests and report back.

I did notice that when doing these tests the CPU gets throttled due to the CPU temperature being above the threshold, which I guess is normal for this type of programs.

MrChromebox, 2020-12-01:
I can provide a test firmware which doesn't set the TDP values as high (ie, back to stock power levels)

lanrat, 2020-12-01:
How much lower are they set in your firmware vs. stock?

I'm using the OEM cooling which I'm assuming is not the best.

MrChromebox, 2020-12-01:
stock is 15W/28W for PL1/2, current UEFI is 28W/51W

lanrat, 2020-12-03:
Would the TDP values cause the entire system to freeze?

I would assume they would just down-clock the CPU and not cause a lock-up.

MrChromebox, 2020-12-03:
hard to say, but I'm not able to reproduce here on a Celeron box

dixonalistair85, 2020-12-12:
Hi - I have a similar issue with my ASUS Chromebox 3 (CN65) - any chance of the test firmware with lower TDP values to see if that helps?

CageOff, 2021-12-23:
Hello everyone!
Can someone email me a BIOS DUMP from CN65 i7-8550u?
My BIOS is damaged. I want to bring him back to life.
Thanks!
CageOff@mail.ru

MrChromebox, 2021-12-24:
@CageOff please do not hijack this issue. Also, see https://wiki.mrchromebox.tech/Unbrcking for important info on directly flashing Chromeboxes.

dgranz, 2022-04-05:
Has this been solved? I have 3 Chromebox3/i7-8550U CPU machines that are having the same stability issues when under load.

bam80, 2022-04-05:
Do we even know it's firmware related? Could we try to reproduce it with original fw on ChromeOS?

lanrat, 2022-04-05:
I never had any issues with my units when I was running ChromeOS. But I also never used them under as high of a load, or for very long.

egrath, 2022-04-05:
> I could never get my initial CN65 (Chromebox3) working under high load without crashing. However I've since had good luck on some other units that have been stable for a few months. So it might be hardware related? Maybe a slightly different revision that causes problems with this firmware?

The original Firmware sets the CPU TDP Limit to 15 W so it's always inside a very safe and conservative margin compared to the 28 W set by MrChromebox's Firmware - Intel recommends a maximum TDP of 25 W to be set by system integrators. IMHO the CN65 cooling system simply can't handle the thermal exhaust when running at 28 W.

bam80, 2022-04-05:
On ChromeOS, we could try to run the same power hungry utils in stock linux container

MrChromebox, 2022-06-01:
> Has the TDP limit been decreased? If so, could you build test fw with 28W/51W please?

it's 20/40 as I mentioned above. I never had any issues with 28/51 so that's what I use myself

bam80, 2022-06-01:
@MrChromebox nice, could you share 28/51 fw then so I could check if it helps with my issue above?

MrChromebox, 2022-06-02:
there's no way that 20W PL1 is causing throttling when idle.

bam80, 2022-06-02:
> Started to use my Chromebox for real things and faced the opposite problem:
> under high load (compiling), CPU clock is maxed to 1.8GHz despite "Frequency Boost" setting with Ondemand governor. CPU temp is about 60-64C all this time, with 36C when idle.

I tried cold boot (from unpowered state) and it fixed the problem for me. Usual reboot didn't fix it.
I didn't try to shut down/switch on system while connected to power.
So it still might be something with EC I think. I'll report it separately when I reproduce it again.

klam2003, 2024-03-16:
I know this is an old post, but I'm still having this exact issue in 2024. I'm using an Asus Chromebox 3 i7-8550U to run a Minecraft server and it crashes when power usage goes above ~25W. Has anyone found a solution to this issue? Cold booting didn't change anything for me. 

I'm creating a Github account just to comment on this, so let me know if there are any additional logs I need to upload to help resolve this issue.

bam80, 2024-03-16:
@MrChromebox could it be helpful to investigate the problem with self-made Suzy-Q cable?
If so I could share my experience of making one from just a pair of resistors and USB-C breakout board.

bam80, 2024-05-13:
> I found couple of reddit posts referring to the same problem.

@aiac 
Could you post the links here for the reference?

aiac, 2024-06-12:
> > I found couple of reddit posts referring to the same problem.
> 
> @aiac Could you post the links here for the reference?

Now I can't find every instance of comments where someone has complained about this issue, but here are two examples:
[https://www.reddit.com/r/chrultrabook/comments/spl990/issues_when_running_windows_10_on_cn65](https://www.reddit.com/r/chrultrabook/comments/spl990/issues_when_running_windows_10_on_cn65)
[https://www.reddit.com/r/chrultrabook/comments/hwc5ng/experiences_from_an_asus_chromebox_3_cn65](https://www.reddit.com/r/chrultrabook/comments/hwc5ng/experiences_from_an_asus_chromebox_3_cn65)

I would be grateful for a new firmware with original PLs, I probably won't be able to compile it myself.

The ability to configure these limits in software would be best because we could test the settings on specific devices.

MrChromebox, 2024-06-13:
> The ability to configure these limits in software would be best because we could test the settings on specific devices.

once the necessary framework exists under coreboot and edk2 I'll be happy to do that, but for now recompilation is the only way. The PLs in the latest release (4.22.5) are very close to stock and should not be problematic

aiac, 2024-06-13:
> once the necessary framework exists under coreboot and edk2 I'll be happy to do that, but for now recompilation is the only way. The PLs in the latest release (4.22.5) are very close to stock and should not be problematic

Is this version already available for updating from your script? I couldn't find a changelog for it anywhere. If there is a way to install it, I will be happy to test it. Thanks very much for your work!

MrChromebox, 2024-06-13:
> Is this version already available for updating from your script? I couldn't find a changelog for it anywhere. If there is a way to install it, I will be happy to test it. Thanks very much for your work!

yes, it is - running the script tells you if there's an update available

blackjid, 2024-06-13:
I can't find any reference to `4.22.5`, in the scripts repository the latest commit that makes reference to a version is `4.22.4`.
The script only upgrades to 4.22.4

```
Downloading Full ROM firmware
(coreboot_edk2-fizz-mrchromebox_20240416.rom)
```

```
**   Device: Asus Chromebox 3 / CN65 (TEEMO)
** Platform: Intel KabyLake
**  Fw Type: Full ROM / UEFI (pending reboot)
**   Fw Ver: MrChromebox-4.22.4 (04/16/2024)
**    Fw WP: Disabled
```

MrChromebox, 2024-06-13:
typo, I meant 4.22.4

aiac, 2024-06-14:
> typo, I meant 4.22.4

I have had this version for some time and the problem remains. the processor exceeds 90 C, and a moment later Windows turns off due to a Thermal Event. Only if I turn on ThrottleStop -> Shift EPP / PL1 = 15, PL2 = 30 does this stop happening and the temperature does not exceed 90 C.

MrChromebox, 2024-06-14:
what's the fan speed? wonder if it's not ramping up enough. The next release will have an EC update for Fizz and I'm running the current PLs (20/40) with Prime95, 4k video, etc and temps don't get above about 65*C

aiac, 2024-06-17:
> what's the fan speed? wonder if it's not ramping up enough. The next release will have an EC update for Fizz and I'm running the current PLs (20/40) with Prime95, 4k video, etc and temps don't get above about 65*C

Where can I check fan's RPM? HwInfo doesnt show them. Based on my ears, I can assume that the fan is running as fast as it can :)

Try running Prime95 and CrystalDiskMark at the same time. In my case, with PL20/40, the computer turns off after simpler tasks, such as reading and analyzing songs from the disk in a DJ program (NI Traktor).

aiac, 2024-09-03:
> try using: https://github.com/death7654/Chrultrabook-Tools

Thank you, it seems I've found the cause, but it also appears to be related to the software. If the fan runs at MAX, it manages to lower the temperature below 90°C even under heavy load.

-   Chrultrabook-Tools shows temperatures up to 56°C
-   At the same time, HwInfo and ThrottleStop show CPU temperatures around 90°C
-   (At 52°C in this program, the CPU temperature is around 75°C)
-   If I set the fan speed to 100% on the curve at 56°C, it actually ramps up under load, allowing the processor to cool down.

Does this mean that the software from EC or something else is incorrectly reading the CPU temperature, leading to improper fan speed settings? What can I do about this?

If I set the speed to AUTO in this tool, the temperature reaches 56°C, the fan spins at 60%, and then the system shuts down due to a thermal event.

By the way, if I want to change PL1 and PL2 and compile the firmware, should I change the values in the `coreboot\src\mainboard\google\fizz\variants\baseboard\include\baseboard\acpi\dptf.asl` file or somewhere else?

MrChromebox, 2024-09-03:
> By the way, if I want to change PL1 and PL2 and compile the firmware, should I change the values in the `src\mainboard\google\fizz\variants\baseboard\include\baseboard\acpi\dptf.asl` file or somewhere else?

no, you set them in `src/mainboard/google/fizz/mainboard.c` in `mainboard_set_power_limits()` - there's a section I added there specifically for the i7 SKU.

aiac, 2024-09-04:
I did some more testing, and on my device, Windows shuts down under heavy load, even if the fan is set to 100% all the time. So, the last thing I can do is lower the PLs. I managed to compile the firmware using Docker. Can you estimate how likely it is that I will brick the device after flashing firmware compiled from the default MrChromebox-2405 branch?

Dockerfile:
```
FROM ubuntu:22.04

RUN apt-get update
RUN apt-get install -y bison build-essential curl flex git gnat imagemagick libncurses5-dev libssl-dev m4 zlib1g-dev pkg-config nasm python3 uuid-dev

VOLUME ["/coreboot"]
WORKDIR /coreboot
```

Rest of the recipe:
```
docker build --tag 'aiac/compile-coreboot-mrchromebox' .
docker run -it -v "${PWD}\coreboot:/coreboot" aiac/compile-coreboot-mrchromebox bash
ln -s /usr/bin/python3 /usr/bin/python
git clone https://github.com/MrChromebox/coreboot .
# change the PL src/mainboard/google/fizz/mainboard.c in mainboard_set_power_limits()
# if (sku == FIZZ_SKU_ID_I7_U42) {
#	conf->tdp_pl1_override = 15;
#	pl2 = 25;
# }
git submodule update --init --checkout --recursive && make crossgcc-i386 CPUS=$(nproc) && ./build-uefi.sh fizz
```

MrChromebox, 2024-09-05:
> I did some more testing, and on my device, Windows shuts down under heavy load, even if the fan is set to 100% all the time. So, the last thing I can do is lower the PLs. I managed to compile the [...]

that's building from upstream coreboot, not my fork. My fork has the PLs set to 20/40, not 15/25

aiac, 2024-09-05:
> that's building from upstream coreboot, not my fork. My fork has the PLs set to 20/40, not 15/25

Yes, the code in comment is already changed to values I would like to try. Im cloning `https://github.com/MrChromebox/coreboot` - the default branch is `MrChromebox-2405` Is this correct repo or it will brick my chromebox? :)

MrChromebox, 2024-09-05:
> Yes, the code in comment is already changed to values I would like to try.

you do know that it's not valid code or valid commenting, right? That will not compile. You can't start a line in a C program with a #.

> Is this correct repo

yes

aiac, 2024-09-05:
> > Yes, the code in comment is already changed to values I would like to try.
> 
> you do know that it's not valid code or valid commenting, right? That will not compile. You can't start a line in a C program with a #.

Yes, thanks, I left the comment for myself between shell commands, so I will remember that I need to stop there and do the change.

> > Is this correct repo
> 
> yes

I didn't have the patience to wait for your confirmation, so I just compiled it and flashed it, crossing my fingers. It worked! The computer is running at full load doing a disk benchmark, and Windows isn't shutting down, everything works perfectly. Thank you so much for your suggestions. I'm glad I don't have to throw the computer away, as I'd rather do that than go back to ChromeOS.

If anyone has a similar problem, they can use the recipe above to compile their own version. For those who are impatient, reckless, and overly trusting, I'm also posting the ready-made file here: [https://aiac.pl/coreboot.rom](url)

MrChromebox, 2024-09-05:
or I can just lower the PLs for the next release...

aiac, 2024-09-05:
> or I can just lower the PLs for the next release...

I don't know how many devices have thermal performance issues. I also don't know how much of a performance gain there is with higher PL1 and PL2 limits. The decision is up to you, of course. In my case, it was unfortunately impossible to work on Windows without spontaneous shutdowns with these limits.

By the way, the link [](https://mrchromebox.tech/files/util/flashrom-libpci38.tar.gz) from the page [](https://docs.mrchromebox.tech/docs/firmware/manual-flashing.html) doesn't work. I ran your update script and used the version of flashrom that it downloads to `/tmp`.

MrChromebox, 2024-09-05:
> By the way, the link from the page doesn't work.

what link from what page?

aiac, 2024-09-06:
[https://docs.mrchromebox.tech/docs/firmware/manual-flashing.html](https://docs.mrchromebox.tech/docs/firmware/manual-flashing.html)

`wget -o flashrom.tar.gz https://mrchromebox.tech/files/util/flashrom-libpci38.tar.gz
`

404

MrChromebox, 2024-09-06:
> https://docs.mrchromebox.tech/docs/firmware/manual-flashing.html
> 
> `wget -o flashrom.tar.gz https://mrchromebox.tech/files/util/flashrom-libpci38.tar.gz `
> 
> 404

fixed

closed 2024-09-06 by MrChromebox

bam80, 2024-09-16:
> or I can just lower the PLs for the next release...

Was it incorporated? If so, could you link the patch here?
Thanks.

MrChromebox, 2024-09-16:
I dropped the previous patch to raise them for the i7-42 SKU

bam80, 2024-09-16:
I asked for the patch link to be sure of the current limitations. As they seem do not documented anywhere..

MrChromebox, 2024-09-17:
I referenced the code location in a comment above:
https://github.com/MrChromebox/firmware/issues/242#issuecomment-2327588301

bam80, 2024-09-17:
> I referenced the code location in a comment above: [#242 (comment)](https://github.com/MrChromebox/firmware/issues/242#issuecomment-2327588301)

> I dropped the previous patch to raise them for the i7-42 SKU

Still don't see the reversion:
https://github.com/MrChromebox/coreboot/commits/next/src/mainboard/google/fizz

Could you mention the issue in the commit next time, please?

MrChromebox, 2024-09-17:
> Could you mention the issue in the commit next time, please?

I can't mention the issue when there is no commit.

Previously, there was a commit to raise/set the PLs for the i7 SKU:
https://github.com/MrChromebox/coreboot/commit/756a96186330f042d08d4c2e1187f32ca99e15e6

that commit was dropped in the 2408 branch

bam80, 2024-09-17:
OK, now I see the commit is specific to MrChromebox-2405 branch.

I thought it was in some sort of "master" branch, so needs to be reverted somewhere.
Thanks for clarifying.

bam80, 2024-09-17:
> > The ability to configure these limits in software would be best because we could test the settings on specific devices.
> 
> once the necessary framework exists under coreboot and edk2 I'll be happy to do that, but for now recompilation is the only way.

That might be recurring problem in the future, could someone file an issue upstream maybe?
I would do myself but do not feel familiar enough with coreboot and edk2 infrastructures.
Thanks.

MrChromebox, 2024-09-17:
> That might be recurring problem in the future, could someone file an issue upstream maybe? I would do myself but do not feel familiar enough with coreboot and edk2 infrastructures. Thanks.

sorry, an issue for what exactly?

bam80, 2024-09-17:
An issue to implement changing PLs at runtime, without recompiling.

MrChromebox, 2024-09-17:
> An issue to implement changing PLs at runtime, without recompiling.

there exists no mechanism to display and modify coreboot settings from edk2. There is one in development, but even after it's complete, one would need to hook up all the settings for a given arch, platform, mainboard, etc. As well as modify coreboot to set given values based on the user settings. it's no small amount of work.

bam80, 2024-09-17:
> there exists no mechanism to display and modify coreboot settings from edk2. There is one in development

So this step would be needed anyway, could you provide a reference to track the development?

MrChromebox, 2024-09-17:
> So this step would be needed anyway, could you provide a reference to track the development?

it's currently being referred to as the Setup Option Table (SOT) feature: https://review.coreboot.org/q/SOT

MrChromebox, 2026-03-23:
in a separate issue, sure