MrChromebox/firmware #243: Upgrade Chromebook soldered RAM possibility?

waitlamp, 2020-12-02:
I'm considering upgrading my Chrome book pro (XE510C24) soldered 4GB RAM.

chrome://system shows that there are 2 (channel?)  Samsung K4E8E324EB_EGCF SO-DIMM.
[image]

For some low-cost Windows tablets, their soldered SDRAM and eMMC can be cheaply replaced by a larger version.
these devices' hardware are very similar to Chromebook. 
(Ultra Low Voltage processor, small soldered RAM, slow soldered eMMC storage etc.)

the key to achieve it is the device have a larger RAM version, or its BIOS support larger SDRAM (though the device have not).

Does UEFI full ROM support a larger SDRAM? (my device have not any larger RAM version)
Could you please tell me some technical details about the Memory and SPD part in the BIOS firmware?

Thank you！

waitlamp, 2020-12-02:
And there is a motherboard picture if it have any help.
[image]

MrChromebox, 2020-12-02:
> Does UEFI full ROM support a larger SDRAM?

it supports all RAM configurations that the stock firmware does, which is to say 2 channels of 2GB Samsung SDRAM, which can be 3 different part numbers.

If you want to replace yours with higher capacity sodimms, you'd need to embed the SPD data for the new modules into the firmware image, then override the mechanism which selects the correct SPD data (both of these are fairly trivial, assuming you have the SPD data for the modules you want to use).

the eMMC should be a little more straightforward, might just require adjusting some tuning parameters in the firmware, or could just work OOTB without modification

waitlamp, 2020-12-02:
> it supports all RAM configurations that the stock firmware does, which is to say 2 channels of 2GB Samsung SDRAM, which can be 3 different part numbers.

3 different part numbers? This information comes for this Firmware? How can I get more information about it?

> If you want to replace yours with higher capacity sodimms, you'd need to embed the SPD data for the new modules into the firmware image, then override the mechanism which selects the correct SPD data (both of these are fairly trivial, assuming you have the SPD data for the modules you want to use).

So, How Can I modify and flash the SPD? Is that a certain flash ROM chip on board that I should use a USB programmer? or I can modify the UEFI firmware image to flash it. What software or hardware should I use to modify it?

Please excuse my ignorance.
Thanks.

MrChromebox, 2020-12-02:
> 3 different part numbers? This information comes for this Firmware? How can I get more information about it?

Assuming this is the Samsung Chromebook Pro (CAROLINE), then:
https://github.com/coreboot/coreboot/blob/master/src/mainboard/google/glados/variants/caroline/Makefile.inc 

> So, how can I modify and flash the SPD? Is that a certain flash ROM chip on board that I should use a USB programmer? or I can modify the UEFI firmware image to flash it. What software or hardware should I use to modify it?

ideally, you would recompile a fresh coreboot/UEFI image, using the SPD hex data for the new modules, and modify the code which selects which SPD hex to load, found here: 
https://github.com/coreboot/coreboot/blob/master/src/mainboard/google/glados/spd/spd.c

> if I copied a SPD information from a similar platform. Would it be available on my device? and what should I do to flash it into UEFI firmware.

ideally, you would use the SPD for the exact DDR sodimm module in use.

> and in some win tablets devices, it's strange that it do not need any configuration to its BIOS or SPD, and it can use many different capacity, different band Soldered SDRAM. What is the principle of it?

it's possible that firmware contains a wide range of SPDs, rather than only the modules in use. It's possible the timings are close enough for the modules substituted in. Most proprietary firmware contains a ton of code not specific to the board/board config. coreboot is the complete opposite in this regard.

waitlamp, 2020-12-02:
> Assuming this is the Samsung Chromebook Pro (CAROLINE), then:
> https://github.com/coreboot/coreboot/blob/master/src/mainboard/google/glados/variants/caroline/Makefile.inc

Thanks for your reply, pardon me I delete my issue earlier, I Searched these part numbers in Samsung website, and I found that they are exactly 3 different capacity SDRAM. 
[image]
Does it means that the its original Firmware can support different capacity sodimms, either the UEFI full ROM (coreboot) ？

MrChromebox, 2020-12-02:
correct -- it would appear that the firmware supports supports 2GB/4GB/8GB configs using those sodimms

waitlamp, 2020-12-02:
> correct -- it would appear that the firmware supports supports 2GB/4GB/8GB configs using those sodimms

That means I don't need to do any configuration or modification to the firmware, the thing I need to do is only replace the  SDRAM?

closed 2020-12-02 by waitlamp

waitlamp, 2021-03-27:
Add：
I have tried replacing the EMMC
The seller replaced the original flash memory with a Samsung EMMC particle with a capacity of 256GB
Samsung EURM4S, although not in a position to disassemble it, I will try to find a more specific model using other software.

I also tried asking to replace 7265AC to AX200
Due to the mismatch of the bottom pads (AX200D2W is 2*4, but the on-board 7265AC is 3*4), the seller did not try flash support for the new version of wifi6 UEFI firmware.

Similarly, perhaps due to the AX200 replacement mistake, the seller did not think it was feasible to replace the memory particles (also considering the reason for the DRAM package pins) and did not do so

I don't think this upgrade service is for everyone, I sent it to Shenzhen. It might be too far for people in other areas
Technically it's not complicated, it's just a hardware EMMC replacement, the problem is that you need to have skilled soldering skills

In theory, using some high speed usb external storage and then booting via usb could also be an alternative to soldering on board EMMC, if the information online is correct, caroline has a 5Gbps usb port, although I have not tested the rate, but if true, it is a less risky way to upgrade storage, but the downside is that the fragile typec connector may be easily damaged

mlukaszek, 2024-07-11:
Sorry to revive an ancient thread, but I'm wondering if you were able to figure out which resistors on the board are responsible for selecting the SPD config? Moving from K4E8E324EB-EGCF (1GB) to K4EBE304EB-EGCG (4GB) would require changing the binary value from `0b0000` to `0b0010` - i.e. moving one of the resistors. Or, patching the [Makefile.mk](https://github.com/coreboot/coreboot/blob/master/src/mainboard/google/glados/variants/caroline/Makefile.mk) to list the modules in a different order, I guess?

theflyingsquirrel88, 2024-11-16:
> Sorry to revive an ancient thread, but I'm wondering if you were able to figure out which resistors on the board are responsible for selecting the SPD config? Moving from K4E8E324EB-EGCF (1GB) to [...]

I am not sure if you are still trying to modify the RAM. I did a mod replacing the EMMC to 128GB and RAM to a total of 16GB successfully.
An alternative to modify the resistor, you can modify the SPI flash. If you are using the mrchromebox UEFI firmware, you can modify it easily.
You can find information in the forum.
https://forum.chrultrabook.com/t/samsung-chromebook-pro-storage-upgrade-ep1/3226
Basically you just replace all RAM SPD to that of the new RAM.
The hardest part I encounter is not modding the SPI flash, but soldering the FBGA153 chips. I did the EMMC without needing to reball, but no matter how many times I redo the RAM chips, I cannot get the chromebook to boot. At the end, I need to reball the chip with low temp 30/70 solder. Then I finally get it to boot.
Just make sure you go from OEM firmware to UEFI following the steps from mrchromebook. I just took a backup of the SPI flash and writing it to the new chromebook and there is always a 30sec boot delay.
So far, I did not come across anyone successfully modifying the RAM. So I think my success is a prove that it can be done.

mlukaszek, 2024-11-17:
That is very impressive, thank you for the detailed write-up. At the moment I put that idea on a back-burner, as the system works fine for what it's being used for, and I wouldn't want to risk breaking it unnecessarily (and no, I did not manage to get mine for $25 :)). I may revisit this in future.