MrChromebox/firmware #244: ASUS Chromebox 3 (CN65) Issues

hellozhangx, 2020-12-07:
Device: ASUS Chromebox 3 (CN65) 
CPU:Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz 
Memorry: 8G DDR4 2400 at DIMM0
Harddisk: 512G NVME SSD
UEFI firmware:coreboot_tiano-fizz-mrchromebox_20200604
Flash the BIOS chip by programer.  

1. USB Failure After Win10 Reboot
The USB port with devices plug-in won't work due to windows reboot.Seems Like they are out of power.
Sleep or Poweroff works just fine.
Tested on Windows 10 Version 1909 and 20H2.
It won't happen on Ubuntu or MacOS.
2. 4K display garbled sometimes
If connecting to 4K display(LG 27UD68W), The boot logo won't display，small lines at the bottom shows instead. After booting windows ,eight little white little rectangles shows up then windows hangs. 
Tested on both  USB type-C to DP caple and HDMI 2.0 caple.
1080P works fine.
3. UHD620(59178086) works fine on macOS 1014.6 but fails to work at macOS 10.15.x or macOS 11.0.1 stack at [IGPU]
same as this guy
https://www.elitemacx86.com/threads/mojave-working-perfectly-catalina-hangs-on-boot-graphics-uhd620-chromebox-i7.525/
I wonder if update GOP vbios may help.

Thanks.

MrChromebox, 2020-12-07:
1) known/documented issue -- not sure how to troubleshoot as Windows-only issue  
2) can't reproduce on my 4K TV, don't have a 4K monitor to test
3) no ability to test/debug MacOS  

I can try updating the UEFI GOP driver, send me an email at mrchromebox at gmail and I'll whip up a test build

edmundasa, 2021-07-30:
@MrChromebox Hi, I have same issue with USB power not available after restart. Could you point me in right direction on how can I troubleshoot it? Could I provide some logs to you to.look over? Maybe RDP connection to machime in queation would help?
Thanks.

edmundasa, 2021-07-30:
> My CN65 never works with my 4k monitor as well. It only works with my 1080p screen.

Hi, I just installed windows 10 couple days ago, I have LG 4k display, works no problem. Maybe you can try update to latest firmware/wimdows update?

MrChromebox, 2021-07-30:
> @MrChromebox Hi, I have same issue with USB power not available after restart. Could you point me in right direction on how can I troubleshoot it? Could I provide some logs to you to.look over? Maybe RDP connection to machime in queation would help?
> Thanks.

I have no idea how to debug this issue, I'm not a Windows dev/user

richb93, 2021-10-03:
Hello, I think I've found a little more information which would be pertinent to fixing this issue. If a reset is issued to the USB Root Hub, any devices that are plugged into the ports die - I assume that Windows is issuing a reset to the Root Hub, causing this behaviour (I invoked it manually using a USB device tree application).

My assumption is that there is a problem in the firmware where the USB Root Hub does not successfully re-initialise following a reset.

I have one of these and a TL866 programmer so would be happy to be a guinea pig for any potential fixes.

MrChromebox, 2021-10-04:
> I invoked it manually using a USB device tree application

from Linux or Windows? If you could manually invoke it from Windows, then shutdown/power back on, would be curious to see if the result was the same as rebooting or not

MrChromebox, 2021-10-06:
> Ah, my apologies. I believe that what you've described is the case- issuing the reset and rebooting causes the same issue.

but does issuing the reset and then shutting down/powering back on show the same behavior?

richb93, 2022-04-30:
Hello - me again - thanks again for your continued efforts to open these devices up.

Regarding point 3 on OPs original list, I have just set up macOS (12.3) on my Chromebox and can repro the issue - from what I've seen online, there is potentially a bug in Coreboot and how it sets up(?) the UHD 620 - the system gets to a certain point at boot and then hard locks during GPU init. From what I've seen online, someone who has installed Coreboot on their Thinkpad with the same KBL-R / UHD620 graphics chip has the same issue, hence my reasoning that the issue is at a Coreboot level.

Interestingly, this issue does not happen on older versions of macOS (10.14), which could be a useful in booting a system up and retrieving information on how the GPU is configured.

From what I can tell, the failure happens just before WhateverGreen (the macOS kernel extension that patches macOS video drivers to work with GPUs on non-Apple devices) tries to start prodding and poking the GPU firmware.

I can provide debug logs from a debug version of the WhateverGreen kernel extension to see if it'll help with what's happening. I do absolutely appreciate however that you have no interest in macOS, or currently a setup yourself to reproduce the issues. Therefore, if you do not wish to investigate further, I totally understand - I thought I'd at least leave this message here for completeness.

MrChromebox, 2022-04-30:
if you want to pastebin some logs and link here I'll take a look, but that's about the best I can do. There is a Hackintosh channel on the chrultrabook Discord server, so you might try asking in there