MrChromebox/firmware #245: CN62     how to increase DVMT on CN62

czh6888, 2020-12-09:
How to increase DVMT on CN62 , I found that 96MB DVMT not enough smooth 4K monitor in macos. How to increase DVMT.

MrChromebox, 2020-12-09:
how did you come to the conclusion that the DVMT is the issue?

it's not configurable, even in firmware. I had to hex-edit the MRC (ram init) blob to modify it

czh6888, 2020-12-14:
framebuffer-stolenmem + framebuffer-fbmem + framebuffer-cursormem（Haswell Only）should less than DVMT.
    If framebuffer-stolenmem + framebuffer-fbmem less than 64MB in HD5500 , Macos cannot recognize 4K 3840*2160.
It only can recognize 2560*1440.  
   Now I set that framebuffer-stolenmem=64MB framebuffer-fbmem=31MB, Macos  can recognize 3840*2160. But I found that not enough smooth in Macos.
   When I set framebuffer-stolenmem + framebuffer-fbmem > 96MB, Macos Crash. So I think that DVMT in coreboot is 96MB
   But when I turn into Windows10, I don't know why Windows10 recognize DVMT 128MB. And Windows10 is smooth enough. So I think maybe increase DVMT can make Macos smooth.

MrChromebox, 2020-12-14:
the DVMT is set to 96MB for all HSW/BDW devices in my UEFI firmware; I can't speculate as to why Windows reports differently or runs more smoothly.

closed 2024-04-28 by MrChromebox