MrChromebox/firmware #253: 2GB Ram on Newest Update

Lrdsnow, 2021-02-09:
Windows 10 on Asus C202SA.
Windows says 2 GB's of ram 
CPU-Z says 2 GB's of ram 
Bios/Firmware says 2 GB's of ram
 
Chromebook C202SA has 4GB's of ram pls help

MrChromebox, 2021-02-09:
the firmware reads the RAM config directly from the board itself, nothing has changed with that code in years, so I suspect it is not misreporting things

Lrdsnow, 2021-02-09:
Then Why Did Raspberry pi os in SeaBios read 4GB just yesterday.

MrChromebox, 2021-02-09:
good question, I'd need to compare the firmware boot logs from both to see why it's being misidentified

Lrdsnow, 2021-02-09:
how can i find those?

MrChromebox, 2021-02-10:
boot Linux USB. download and extract: https://mrchromebox.tech/files/util/cbmem.tar.gz

then run: `sudo ./cbmem -1 > cbmem.log` (name them uniquely) and throw on pastebin or email to me at MrChromebox at gmail. Do for both UEFI and stock/RW_LEGACY ideally, but if you want to just do UEFI first I'll take a look and see if anything obvious

closed 2021-03-16 by Lrdsnow